MarlinFirmware/Marlin #26583: NO COMPILATION

NiKoLo2002, 2023-12-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

when i start the compiling sequence on VS marlin. i get these error

PS C:\Users\nicolò\Desktop\FIRMWARE HYPERCUBE> platformio run --silent -e STM32G0B1RE_btt
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open map file C:/Users/nicolò/Desktop/FIRMWARE HYPERCUBE/.pio/build/STM32G0B1RE_btt/firmware.map: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32G0B1RE_btt\firmware.elf] Error 1
================================================================================================== [FAILED] Took 13.74 seconds ==================================================================================================

Environment      Status    Duration
---------------  --------  ------------
STM32G0B1RE_btt  FAILED    00:00:13.740
============================================================================================= 1 failed, 0 succeeded in 00:00:13.740 =============================================================================================

### Expected behavior

i expect to recive the bin file

### Version of Marlin Firmware

02010201 - last

### Printer model

hypercube - autobuilt

### Electronics

SKR-MINI-E3-V3.0

### LCD/Controller

BTT E3 TFT3.5

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13786444/Configuration.zip)

thisiskeithb, 2023-12-28:
Your username / file path cannot have special characters (`ò`), so you’ll need to move Marlin to another location (like `C:\marlin`).

closed 2023-12-28 by thisiskeithb

NiKoLo2002, 2023-12-28:
ok i try

NiKoLo2002, 2023-12-28:
@thisiskeithb i tryed and i recive that error

PS C:\FIRMWARE HYPERCUBE> platformio run --silent -e STM32G0B1RE_btt
platformio : Termine 'platformio' non riconosciuto come nome di cmdlet, funzione, programma eseguibile o file script. Controllare l'ortografia del nome o verificare che il percorso sia incluso e corretto, quindi riprovare.   
In riga:1 car:1
+ platformio run --silent -e STM32G0B1RE_btt
+ ~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (platformio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

EvilGremlin, 2023-12-28:
Keith was a bit incorrect. You have to change username to valid one (create new user).

NiKoLo2002, 2023-12-28:
> Keith was a bit incorrect. You have to change username to valid one (create new user).

I tryed with my desktop PC that have the dame setting of the laptop and only One errore that was about spi, i change something and all worked. On the laptop i had that problem After some class of python because before that all was working

thisiskeithb, 2023-12-28:
> Keith was a bit incorrect. You have to change username to valid one (create new user).

I've amended my comment to clarify that includes creating a new user account. You cannot have special characters in the path.

> On the laptop i had that problem After some class of python because before that all was working

Try cleaning your PIO install: https://marlinfw.org/docs/basics/troubleshooting.html#platformio-errors

1. Delete the hidden `.pio` folder at the root of your Marlin project.

2. Delete the hidden `.platformio` folder from your User directory:
 
   - Windows: `%USERPROFILE%\.platformio\`

3. Reinstall PlatformIO.

NiKoLo2002, 2023-12-28:
Ok

NiKoLo2002, 2023-12-28:
@thisiskeithb now work all fine thanks also with the ò letter