MarlinFirmware/Marlin #26587: [BUG] Can't build while #define CR10_STOCKDISPLAY is uncommented (latest bugfix 2.1.x)

PierreGillain, 2023-12-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Good morning, i tried to update my Ender 3 to add BL_Touch support. 
I use the last bugfix 2.1.x firmware and change all the parameters in configuration.h to be compatible with my Ender 3 and my BL Touch.
I can compile (with Auto Build Marlin) my firmware only when i comment out #define CR10_STOCKDISPLAY, after that i was able to upload my firmware (it worked, i check with printrun software) but the screen stay empty (quite logical because i didn't define any screen).
I saw that some people had the same issue in the past with july 2 update but i can't find a fix for the current version.

Thank you for helping me, I hope I made myself understood despite my poor level of English.

### Bug Timeline

28/12/2023

### Expected behavior

The firmware to compile with #define CR10_STOCKDISPLAY uncommented

### Actual behavior

it doesn't compile
[config.zip](https://github.com/MarlinFirmware/Marlin/files/13790880/config.zip)

### Version of Marlin Firmware

2.1.x

### Printer model

Creality Ender-3

### Electronics

1.1.5 motherboard

### LCD/Controller

ENH12864Z

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-12-29:
No steps to reproduce

no error messages

This could be due to recent u8glib bug, or it could be because firmware no longer fits in 128k of flash in the decade old MELZI_CREALITY

PierreGillain, 2023-12-29:
Thanks for answering, do you know what can I do to fix it ? Should i change melzi for something else or should i tried with an old specific Marlin bugfix version ?

plampix, 2024-01-02:
I can add some more PRs to reduce memory usage some more. Will do that today or tomorrow.

PierreGillain, 2024-01-07:
It was only about memory usage so i finally build a firmware by commenting out some features and optimizing compilation

closed 2024-01-07 by PierreGillain