MarlinFirmware/Marlin #26589: [BUG] G28 w Sensorless homing + Bltouch m999 error

Eno423, 2023-12-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

new build from 2.1 bugfix from Dec 23rd 2023

Enabled sensorless homing on X and Y

Now when running G28 it homes X & Y w/o issue, but if my Z is more then 3-5mm off the build plate the bltouch probes in mid air & throws a m999

this did not happen on the Jan 2023 bugfix 2.1 I was running previously / before I enabled sensorless homing 

### Bug Timeline

new, 5 days ago

### Expected behavior

I expect G28 to home X and Y & drop Z down as far as it needs to until the BLtouch triggers

### Actual behavior

it homes X & Y, & then the z drops about 3-5 mm & triggers in mid air a few times & then throws a m999

### Steps to Reproduce

run g28
watch X home
watch Y home
Watch Z move down a few mm & then blotuch probes in air & fails

### Version of Marlin Firmware

Bugfix 2.1 (Dec 23, 2023)

### Printer model

umm..Pru-bear-custom?

### Electronics

SKR mini e3 v2

### LCD/Controller

Btt tft 3.5 e3 v3

### Other add-ons

Bltouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/13790965/Configuration_adv.zip)
[G28_error.txt](https://github.com/MarlinFirmware/Marlin/files/13790973/G28_error.txt)

Injineers, 2023-12-30:
[G28_error.txt](https://github.com/MarlinFirmware/Marlin/files/13796960/G28_error.txt)

Injineers, 2023-12-30:
new log, this time 

manually jogg'ed Z down to about 3-5mm off the bed (terminal thought it was -63 mm on z) 

ran g28, Z lifted up to around where it originally was on boot, then dropped back down to the bed & probed like normal
[g28_success_but_odd.txt](https://github.com/MarlinFirmware/Marlin/files/13796983/g28_success_but_odd.txt)

closed 2024-01-01 by Injineers

thisiskeithb, 2024-01-01:
To close things out, this was a wiring issue.