MarlinFirmware/Marlin #26590: [BUG] E0_AUTO_FAN_PIN sanity check for FAN0_PIN is not triggred

ellensp, 2023-12-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Take stock Configs, set #define E0_AUTO_FAN_PIN FAN0_PIN
Watch as Marlin compiles without any issues
despite having the following in sanity check.
```
    #if PIN_EXISTS(E0_AUTO_FAN) && E0_AUTO_FAN_PIN == FAN0_PIN
      #error "You cannot set E0_AUTO_FAN_PIN equal to FAN0_PIN."

```

 

### Bug Timeline

Unknown

### Expected behavior

This should trigger the error in sanity check "You cannot set E0_AUTO_FAN_PIN equal to FAN0_PIN."

### Actual behavior

no error

### Steps to Reproduce

1. set #define E0_AUTO_FAN_PIN FAN0_PIN
2. compile
3. note it compiles without error

### Version of Marlin Firmware

Bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-12-30:
In Marlin/src/inc/SanityCheck.h is the following code

```CPP
#if HAS_AUTO_FAN
  #if HAS_FAN0
    #if PIN_EXISTS(E0_AUTO_FAN) && E0_AUTO_FAN_PIN == FAN0_PIN
      #error "You cannot set E0_AUTO_FAN_PIN equal to FAN0_PIN."
    #elif PIN_EXISTS(E1_AUTO_FAN) && E1_AUTO_FAN_PIN == FAN0_PIN
      #error "You cannot set E1_AUTO_FAN_PIN equal to FAN0_PIN."
    #elif PIN_EXISTS(E2_AUTO_FAN) && E2_AUTO_FAN_PIN == FAN0_PIN
      #error "You cannot set E2_AUTO_FAN_PIN equal to FAN0_PIN."
    #elif PIN_EXISTS(E3_AUTO_FAN) &&  E3_AUTO_FAN_PIN == FAN0_PIN
      #error "You cannot set E3_AUTO_FAN_PIN equal to FAN0_PIN."
    #endif
  #endif
#endif
```

But HAS_FAN0 is not set, so this block is ignored.

In Marlin/src/inc/Conditionals_post.h has this code

```CPP
#define _IS_E_AUTO(N,F) (PIN_EXISTS(E##N##_AUTO_FAN) && E##N##_AUTO_FAN_PIN == FAN##F##_PIN)
#define _HAS_FAN(F) (F < MAX_FANS && PIN_EXISTS(FAN##F) \
                     && !(HAS_CONTROLLER_FAN && CONTROLLER_FAN_PIN == FAN##F##_PIN) \
                     && !_IS_E_AUTO(0,F) && !_IS_E_AUTO(1,F) \
                     && !_IS_E_AUTO(2,F) && !_IS_E_AUTO(3,F) \
                     && !_IS_E_AUTO(4,F) && !_IS_E_AUTO(5,F) \
                     && !_IS_E_AUTO(6,F) && !_IS_E_AUTO(7,F))

#if _HAS_FAN(0)
  #define HAS_FAN0 1
#endif
```

The test explicitly checks for E0_AUTO_FAN == FAN0_PIN and tells marlin that HAS_FAN0 is false, disabling the sanity check.

ellensp, 2023-12-30:
proposal, treat FAN0 as special, only check it has been defined. 

```DIFF
diff --git a/Marlin/src/inc/Conditionals_post.h b/Marlin/src/inc/Conditionals_post.h
index 04e96177b5..d946904e75 100644
--- a/Marlin/src/inc/Conditionals_post.h
+++ b/Marlin/src/inc/Conditionals_post.h
@@ -2657,7 +2657,7 @@
                      && !_IS_E_AUTO(4,F) && !_IS_E_AUTO(5,F) \
                      && !_IS_E_AUTO(6,F) && !_IS_E_AUTO(7,F))
 
-#if _HAS_FAN(0)
+#if PIN_EXISTS(FAN0)
   #define HAS_FAN0 1
 #endif
 #if _HAS_FAN(1)

```

thisiskeithb, 2023-12-30:
> proposal, treat FAN0 as special, only check it has been defined. as it was originally

Can confirm that the sanity check now fails when it should.

ellensp, 2023-12-30:
PR created, closing.

closed 2023-12-30 by ellensp