MarlinFirmware/Marlin #26592: Fails to home when FT Motion is enabled

ahmetcemturan, 2023-12-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If FT Motion is Enabled the Y-Endstop (Y-MAX) is not working.

### Bug Timeline

New Bug. Must have started between Oct 13th and Dec 27th

### Expected behavior

I expected it to home correctly.

### Actual behavior

Fails to home correctly.
I have  Quickhome Enabled: if X Hits first then it will not continue Y-movement, and move on to lower Z, omittitting to move to the center of the build plate for that. (let it never complete as the probe would not be triggered as its off the printplate.
If Y hits first it will just ignore it and keep moving.

IF I disable FT Motion either in FW or via G-Code it homes fine.

Reboot FT OFF  > Homing OK  > Movement OK > FT On > Movement OK > Homing Fail

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.1.x (Dec 30 2023 09:31:12)

### Printer model

Sapphire Pro

### Electronics

Robin Nano S 1.3

### LCD/Controller

MKS_ROBIN_TFT35

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin 30-12-2023.zip](https://github.com/MarlinFirmware/Marlin/files/13797503/Marlin.30-12-2023.zip)

narno2202, 2023-12-30:
@ahmetcemturan , i can confirm the bug with latest bugfix. As i was running a previous version on my printer all was fine.
bugfix  rev aa4aa7c3d6eab30cb49f93330df7eeac0d73aa8e is working fine on my repo (as i remove the original branch, i don't know if you can try it, this is the commit with sensorless homing fix) . I'm currently investigating the issue.

ahmetcemturan, 2023-12-30:
I have tried to run that one with FT motion disabled and tried to print.  About 1mm into the print I believe Z layer shifted.  Could have been Marlin or not.  Did not pursue it further and reverted to my old working version though.  So maybe you can reproduce it?

ahmetcemturan, 2023-12-30:
I guess this is the one:   https://github.com/MarlinFirmware/Marlin/tree/fe8266b59105c24a6dced92bfd604374e5e9ef8d
I will try it.

thisiskeithb, 2023-12-30:
@ahmetcemturan: Please confirm that https://github.com/MarlinFirmware/Marlin/pull/26595 fixes your homing issue.

ahmetcemturan, 2023-12-31:
I am not sure about the Commit number but I am using the version of the link above without problems.
Homşng issue is fixed.

ahmetcemturan, 2023-12-31:
I understand that 26595 is a new one.  
I will try it out and report back.

closed 2023-12-31 by thinkyhead

ahmetcemturan, 2024-01-09:
I was able to try the latest commit now:
The latest commit still wont home correctly.