MarlinFirmware/Marlin #26593: [BUG] Simulator build failure since commit #9d324f7 (27 december)

narno2202, 2023-12-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Simulator build fails with 'usleep not declared in this scope'. Definition removed in 27 december commit #9d324f7.
Solved by reverting this commit

### Version of Marlin Firmware

latest bugfix

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-12-30:
tested with 
simulator_linux_debug + https://github.com/MarlinFirmware/Marlin/pull/26532 
simulator_linux_release + https://github.com/MarlinFirmware/Marlin/pull/26532
linux_native

All built fine

What operating system are you using?

ellensp, 2023-12-30:
Confirmed usleep is used in U8glib-HAL

https://github.com/MarlinFirmware/U8glib-HAL/blob/master/src/clib/u8g_delay.cpp#L106C14-L110

ellensp, 2023-12-30:
restored https://github.com/MarlinFirmware/Marlin/commit/8f63a0a01fadf676e5328a86b350787c251fbe1b

closed 2023-12-30 by ellensp

thinkyhead, 2023-12-30:
reverted!