MarlinFirmware/Marlin #26594: [BUG] path truncated during build for MKS Tinybee

ahmetcemturan, 2023-12-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When trrying to compile the latest (30-12) build I get the error:

In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/command.cpp:21:0:
.pio/libdeps/mks_tinybee/ESP3DLib/src/esp3dlibconfig.h:45:44: fatal error: D:/Users/Ahmet/Documents/NO/inc/MarlinConfigPre.h: No such file or directory
compilation terminated.

### Bug Timeline

It is existent in the latest bugfix

### Expected behavior

I expect to compile correctly

### Actual behavior

It fails to compile giving this error:  
In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/command.cpp:21:0:
.pio/libdeps/mks_tinybee/ESP3DLib/src/esp3dlibconfig.h:45:44: fatal error: D:/Users/Ahmet/Documents/NO/inc/MarlinConfigPre.h: No such file or directory
compilation terminated.

When I copy the directory and contents to that place it skips that error, 
Actual path would be...  D:\Users\Ahmet\Documents\NO GIT\MKS\TinyBee\30-12-23\Marlin\src

### Version of Marlin Firmware

2.1.bugfix as of December 30

### Printer model

Prusa Clone

### Electronics

Tinybee

### LCD/Controller

Reprap Smart

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13797774/Marlin.zip)

ellensp, 2024-01-02:
1) Windows has limits on its path lengths. It is advised to move Marlin to the root of a drive
2) Various parts of platformio and the GCC tools it uses do not handle spaces or special characters in the path. Remove all spaces and non English characters from the path.

Both of these issues are outside of Marlin

closed 2024-01-02 by ellensp

ahmetcemturan, 2024-01-03:
[image]
[image]
Indeed it compiles when I copy the folder into Root BUT:
I have both of these folders in the same folder.  One compiles, the other one doesn't.  Also I can compile the same release for Robin Nano.. (about the same length of folder names maybe even a bit longer..)

So just FYI.

Thanks

thisiskeithb, 2024-01-03:
When we say copy files to the root of your drive, we mean like `C:\marlin\` without any additional subfolders.

Your screenshot shows the `Documents` folder which is generally `C:\Users\<username>\`, plus you have additional subfolders, so you are not in the root of your drive. 

That file path should look like the following:

[image]

ahmetcemturan, 2024-01-03:
Yeah of course... I use the boardname + release date instead of "Marlin"..  It compiled dont worry but as I said above... weird...   (as most things with Tinybee probably)