MarlinFirmware/Marlin #26600: [BUG] M48 calling G28 crashes Z into bed.

Tannoo, 2023-12-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

M48 crashes Z into the bed when it is homing before the probe test.

### Bug Timeline

Noticed when setting up and testing a BLTouch for the first time.

### Expected behavior

I expected it to home properly like it does every other time.

### Actual behavior

With the printer unhomed.
M48 sees the printer is not homed and initiates homing.
X and Y home just fine then Z starts homing, but ignores the Z endstop.

Probe is out of the question due to it not setup to be the Z endstop.

### Steps to Reproduce

Turn on the printer. Or disarm the steppers.
Send M48 for a probe test.
If your machine uses a separate Z endstop, it will crash as it is ignored.
If your machine uses the probe for the Z endstop, I am sure it will be fine.

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Dec 31 2023 09:12:16)

### Printer model

BTT-Outfitted Kobra Max

### Electronics

BTT SKR 3 EZ

### LCD/Controller

BTT TFT35 E3 V3.0.1

### Other add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/13801841/Configs.zip)

thisiskeithb, 2023-12-31:
Let's gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun, Repetier Host, or OctoPrint.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `M48` to run a Probe Repeatability Test.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Tannoo, 2023-12-31:
[Requested log.txt](https://github.com/MarlinFirmware/Marlin/files/13802216/Requested.log.txt)

thisiskeithb, 2023-12-31:
```prolog
Send: M48
Recv: echo:M48
Recv: M48 Z-Probe Repeatability Test
[snip]
Recv: Finished!
Recv: Mean: -2.496350 Min: -2.498 Max: -2.495 Range: 0.003
Recv: Standard Deviation: 0.001246
```

Looks like it worked fine?

Tannoo, 2023-12-31:
M48 runs after M84 without homing.

[M48 runs after M84.txt](https://github.com/MarlinFirmware/Marlin/files/13802224/M48.runs.after.M84.txt)

This is not right.

Next, I will have a crash log where I have to stop the printer.

Tannoo, 2023-12-31:
> ```prolog
> Send: M48
> Recv: echo:M48
> Recv: M48 Z-Probe Repeatability Test
> [snip]
> Recv: Finished!
> Recv: Mean: -2.496350 Min: -2.498 Max: -2.495 Range: 0.003
> Recv: Standard Deviation: 0.001246
> ```
> 
> Looks like it worked fine?

Yes, when you follow those steps. I have stated that also.

Tannoo, 2023-12-31:
Now, I see that M48 no long initiates G28 if it is not homed.

This is good.

thisiskeithb, 2023-12-31:
`M48` will only call `G28` *once* after power up and never again if your machine hasn't homed. It doesn't matter if your steppers end up disabled again or not.

Tannoo, 2023-12-31:
It did not this time, on today's build.

That is okay. Prevents crashes. :)

Tannoo, 2023-12-31:
Thank you. Someone changed something from yesterday's build. I am not digging into it. But, it is satisfactory now.

closed 2023-12-31 by Tannoo

thisiskeithb, 2023-12-31:
Nothing related to the BLTouch/homing has changed, but glad it is no longer an issue.

thisiskeithb, 2023-12-31:
I would highly recommend removing your Z endstop and use your probe for Z homing. It'll simplify the entire first layer calibration process.

Tannoo, 2023-12-31:
> I would highly recommend removing your Z endstop and use your probe for Z homing. It'll simplify the entire first layer calibration process.

It has been a thought, but I have also never used a BLTouch before.

So, on this machine rebuild, I was using what I know works for now.
My other printer uses the probe for the Z endstop.

Sensorless homing is on the list before removing the Z endstop yet.

Tannoo, 2023-12-31:
Nevermind, I found it.

Tannoo, 2023-12-31:
> Nothing related to the BLTouch/homing has changed, but glad it is no longer an issue.

Now, I have to check in to a looney farm.

Tannoo, 2024-01-01:
I am going to re-test this. The issue was initiating G28 after M48 fails with the unhomed error.