MarlinFirmware/Marlin #26601: Marlin (LTS/Bugfix) down't compile on Arduino IDE(2.2.1) onUbuntu(22.04.3)

mlueft, 2024-01-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin (Bugfix and LTS) doesn't compile in Arduino IDE (2.2.1) on Ubuntu (22.04.3 LTS)

### Steps to Reproduce

1. Clone Marlin from Github.
2. In Arduino IDE:
2.1. download/install extension for STM32Fxx/GD32F1XX
2.2. Set Board to 'Generic STM32F103R'
2.3. CPU speed to '72 Mhz (Normal)'. Default, I didn't change it.
2.4. Variant: "STM32F1033R8 (20k RAM. 64k Flash)". Default, I didn't change it.
2.5. Optimize: "Smallest /default)". Default, I didn't change it.
2.6. Upload method: "STM32duino bootloader". Default, I didn't change it.
3. In Configuration.h to:
3.1. Set Board to BOARD_CREALITY_V427.
3.2. Activate FAN_SOFT_PWM. A Compiler message requests this. (Two years ago I used Used Marlin with this board, there it is deactivated.)
4. Varify code.

### Version of Marlin Firmware

Bugfix and LTS

### Printer model

Orca 0.42

### Electronics

Creality Silent board 4.2.7

### LCD/Controller

MKS H43

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp: In function 'void waitSpiTxEnd(spi_dev*)':
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp:97:20: error: redefinition of 'void waitSpiTxEnd(spi_dev*)'
 static inline void waitSpiTxEnd(spi_dev *spi_d) {
                    ^
In file included from /home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp:34:0:
/home/work/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:433:20: error: 'void waitSpiTxEnd(spi_dev*)' previously defined here
 static inline void waitSpiTxEnd(spi_dev *spi_d)
                    ^
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp: At global scope:
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp:158:1: error: prototype for 'SPIClass::SPIClass(int8_t, int8_t, int8_t, int8_t)' does not match any in class 'SPIClass'
 SPIClass::SPIClass(int8_t mosi, int8_t miso, int8_t sclk, int8_t ssel) : SPIClass(1) {
 ^
In file included from /home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp:34:0:
/home/work/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:170:7: error: candidates are: constexpr SPIClass::SPIClass(SPIClass&&)
 class SPIClass {
       ^
/home/work/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:170:7: error:                 constexpr SPIClass::SPIClass(const SPIClass&)
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp:105:1: error:                 SPIClass::SPIClass(uint32_t)
 SPIClass::SPIClass(uint32_t spi_num) {
 ^
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp:269:6: error: prototype for 'void SPIClass::beginTransaction(uint8_t, const SPISettings&)' does not match any in class 'SPIClass'
 void SPIClass::beginTransaction(uint8_t pin, const SPISettings &settings) {
      ^
In file included from /home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp:34:0:
/home/work/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:208:7: error: candidates are: void SPIClass::beginTransaction(uint8_t, SPISettings)
  void beginTransaction(uint8_t pin, SPISettings settings);
       ^
/home/work/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:207:7: error:                 void SPIClass::beginTransaction(SPISettings)
  void beginTransaction(SPISettings settings) { beginTransaction(BOARD_SPI_DEFAULT_SS, settings); }
       ^
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp:277:6: error: prototype for 'void SPIClass::beginTransactionSlave(const SPISettings&)' does not match any in class 'SPIClass'
 void SPIClass::beginTransactionSlave(const SPISettings &settings) {
      ^
In file included from /home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp:34:0:
/home/work/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:211:7: error: candidate is: void SPIClass::beginTransactionSlave(SPISettings)
  void beginTransactionSlave(SPISettings settings);
       ^
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp: In member function 'void SPIClass::dmaTransferSet(const void*, void*)':
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/SPI.cpp:386:20: error: 'ff' was not declared in this scope
     transmitBuf = &ff;
                    ^
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/HAL_SPI.cpp: In function 'void spiRead(uint8_t*, uint16_t)':
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/HAL_SPI.cpp:126:54: error: call of overloaded 'dmaTransfer(int, uint8_t*, uint16_t&)' is ambiguous
   SPI.dmaTransfer(0, const_cast<uint8_t*>(buf), nbyte);
                                                      ^
/home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/HAL_SPI.cpp:126:54: note: candidates are:
In file included from /home/work/Documents/Orca/Marlin/Marlin/src/HAL/STM32F1/HAL_SPI.cpp:32:0:
/home/work/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:301:11: note: uint8 SPIClass::dmaTransfer(const void*, void*, uint16)
     uint8 dmaTransfer(const void * transmitBuf, void * receiveBuf, uint16 length);
           ^
/home/work/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:302:11: note: uint8 SPIClass::dmaTransfer(uint16, void*, uint16)
     uint8 dmaTransfer(const uint16 value, void * receiveBuf, uint16 length);
           ^

exit status 1

Compilation error: redefinition of 'void waitSpiTxEnd(spi_dev*)'
'''
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13803866/Configuration.zip)

EvilGremlin, 2024-01-01:
Just don't use arduino, it's REALLY bad. https://marlinfw.org/docs/basics/install_platformio_vscode.html

ellensp, 2024-01-02:
HAL/STM32F1 implies a maple build environment, while you mention STM32duino  which is the alternative to the maple build environment,,  these are not compatible,  STM32duino should use HAL/STM32

And here is the rub with Arduino IDE, you have to manually select and install the correct extensions and libraries including version numbers, while platfromio will do this for you automatically 

Also  "Variant: "STM32F1033R8 (20k RAM. 64k Flash)". Default, I didn't change it."  you should have
Its a STM32F103RCT6 or a STM32F103RET6, definitely not a  STM32F103R8

Please join the 21th century and use platformio under vscode.

mlueft, 2024-01-04:
Hi!

I watched some YT-Videos about compiling marlin. One pointed me in the right direction.
It was mentioned Arduino IDE is for 8-Bit boards and VSC for 32-Bit boards.
Now I remember this, same as two years ago.
Now with VSC and PIO it compiles.
This request can be closed, I think.
Thank you all!

closed 2024-01-04 by mlueft