MarlinFirmware/Marlin #26602: [FR] Auto or manual fan control during PID autotunes

Tannoo, 2024-01-01:
### Is your feature request related to a problem? Please describe.

Controller fan does not operate at all during it's PID autotune.

### Are you looking for hardware support?

I don't believe so.

### Describe the feature you want

I want the controller fan to run during PID autotune. It works in all other instances.

### Additional context

Nope.

classicrocker883, 2024-01-02:
what UI are you using? printer information?

Tannoo, 2024-01-02:
Octoprint

BTT SKR 3 EZ stuffed into an Anycubic Kobra Max.

classicrocker883, 2024-01-02:
have you tried `PID_FAN_SCALING`? check out **Configuration_adv.h**

Tannoo, 2024-01-02:
Yes. I have it enabled.

The controller fan works manually or by heating the bed.

It does not, however, turn on when the bed PID is running. Only AFTER the PID is finished the controller fan comes on.

classicrocker883, 2024-01-04:
for MPC was `MPC_INCLUDE_FAN` or `MPC_FAN_0_ACTIVE_HOTEND`  
enabled?

there may be a setting like that which needs enabling. otherwise I might be able to make it work as a workaround for now by putting in a function in **temperature.cpp**

Tannoo, 2024-01-04:
> set_fan_speed(0, speedyouwant);

No, I have not enabled any MPC functions

Reminder.. there is no issue with my hotend fan now. That was an old issue if the fan being wired backwards and it not working.

This issue is the controller fan that works in all instances other than during PID AUTOTUNE.

I have tried making the fan work in `temperature.cpp`.

I will try this:
```
    #if ENABLED(PRINTER_EVENT_LEDS)
      const celsius_float_t start_temp = GHV(degChamber(), degBed(), degHotend(heater_id));
      LEDColor color = ONHEATINGSTART();
    #endif

    TERN_(TEMP_TUNING_MAINTAIN_FAN, adaptive_fan_slowing = false);

    **set_fan_speed(0,  255);**

    LCD_MESSAGE(MSG_HEATING);

    // PID Tuning loop
    wait_for_heatup = true;
    while (wait_for_heatup) { // Can be interrupted with M108
```

EvilGremlin, 2024-01-04:
This is not a feature request, this is a bug (incomplete implementation). Controller fan supposed to turn on any active heater or motor. Though this is hardly an issue at all, not much heat from MOSFET

Tannoo, 2024-01-04:
My mosfet lives in the controller box and is much bigger than the one on the controller.

The heated bed is a 400x400 and is a 24V heater.

Tannoo, 2024-01-04:
I was told that the controller fan code is fine.

SO

I made this a FR.

Tannoo, 2024-01-04:
During any PID autotune... there are lots of things that are not updated.

The TFT stops getting any info about the printer... no temps, no fan speeds, the only thing is does get is the status messages.
The Octoprint terminal window gets reports on the temperature.... but no fan speeds.

It also doesn't control my controller fan during this time.

The controller fan will come on as soon as the PID is done... but not during.

So..... if the code is fine. I am asking it as a feature request to make it work.

thisiskeithb, 2024-01-04:
> The TFT stops getting any info about the printer... no temps, no fan speeds, the only thing is does get is the status messages.

That is basically expected with serial devices when you run blocking commands. Depending on the TFT firmware (like BigTreeTech's), it will parse & rewrite certain g-codes to be non-blocking, so testing should be done in "Marlin Mode" to bypass any TFT firmware funny business.

EvilGremlin, 2024-01-04:
UI updates doesn't happen because autotune is executed outside regular temp control code. This is also inconsistent behavior which i'd condiser a bug too, but really unimportant one. Not enabling controller fan for PID tune is a bit more of a bug, but you're most likely first person to be bothered about it. Does MOSFET ever get above ~100C? If not - don't bother.

Tannoo, 2024-01-04:
So.... why is the autotune SOO important that it has to block everything?

Printing anything does WAAY more than any PID autotune does.
It doesn't block the NEOPIXELS....
So.... why does it block the controller fan?

Screw the TFT, but spit the numbers out to Octoprint. lol ... nice.

I have no idea if the mosfet is getting to whever temp. I would need to run it on it's side. The controller box is like the Ender 3.

classicrocker883, 2024-01-09:
what about something like this?  
[External MOSFET Controlled Fans](https://www.makenprint.uk/3d-printing/3d-printing-guides/external-mosfet-controlled-fans/)

Tannoo, 2024-01-09:
The issue is not the mosfet. The one on the board works fine.

Tannoo, 2024-01-09:
I have an external mosfet for the bed heater that is cooled by the controller fan.

ellensp, 2024-01-09:
educated guess, haven't tried it  (my test systems are mostly bare boards. no heaters)  

edit Marlin/src/module/temperature.cpp
find 
`    while (wait_for_heatup) { // Can be interrupted with M108
`
add this line under it
`      TERN_(USE_CONTROLLER_FAN, controllerFan.update()); // Check if fan should be turned on to cool stepper drivers down
`

Tannoo, 2024-01-09:
Thanx. I will try that in the AM.

Tannoo, 2024-01-09:
> educated guess, haven't tried it (my test systems are mostly bare boards. no heaters)
> 
> edit Marlin/src/module/temperature.cpp find
> 
> ```c++
>     while (wait_for_heatup) { // Can be interrupted with M108
> ```
> 
> add this line under it
> 
> ```c++
>       TERN_(USE_CONTROLLER_FAN, controllerFan.update()); // Check if fan should be turned on to cool stepper drivers down
> ```

That did it! Thank you!

closed 2024-01-09 by Tannoo