MarlinFirmware/Marlin #26607: Regarding Brushless Motor

s145854, 2024-01-01:
### Describe the feature you want

Marlin firmware - CNC

### Additional context

Hello, so this is how I want to create a CNC machine with brushless motors instead of stepper motors, I have the motors, they also have a driver that I can configure its firmware for, that means I have the source code of the driver

I have hall effect sensors
And also an option for FOC or Six-Step

And I want to use Marlin firmware only I know that Marlin firmware supports stepper motors

The brushless motors I have and the drivers are really high quality and the power of the motor is 40 volts 750 watts

Of course I don't use more than 200 watts

And the motors in a closed loop, most things are already prepared in the motor driver code

The only thing left for me to know is how I can configure the Marlin firmware for the same brushless motor and by the way I have the source code of the driver of the motor

I would be happy to help and I'm even willing to pay money to the Marlin company to contact me and do it in a more private and careful manner, thank you very much!

EvilGremlin, 2024-01-01:
Impossible with any CNC firmware - you must use step/dir interface or write whole new planner/motion code from scratch. So, no matter what your motors are, your drivers must use step/dir input. Closed loop with position encoders is supported, but this code is orphaned and i don't know anyone using it. And i suspect it allow for position overshoot. You're on your own there.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

* MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
* RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
* Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
* Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
* [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
* 3D Printing Discord server. Join link: https://discord.gg/3dprinters
* Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report.

s145854, 2024-01-01:
So you actually mean that it is impossible to use brushless motors instead of stepper motors in Marlin firmware?

When I say brushless motor instead of stepper motor I mean machine axes X, Y, Z And ETC...

EvilGremlin, 2024-01-01:
Well steppers are brushless motors. Point is control type, must be same step pulses. What you actually spin with these pulses in not firmware's buisiness, bit motor driver's one. There are plenty servo drives that se step/dirt input but actual motor is 3-phase BLDC or even AC induction one.

s145854, 2024-01-01:
The type of my motor is 3 Phase BLDC And Voltage is DC

And i have 3 Phase brushless motor driver

I just need to know from you if it will fit me with Marlin firmware or if it doesn't fit that's what I need to know?

EvilGremlin, 2024-01-01:
Again, it's all about motor driver. How's you driver is controlled? If it's not step/dir - then no, not a chance.

s145854, 2024-01-01:
Basically I can edit the pins of the motor driver through software

There is UART data transfer

Still, I'm sending you 2 pictures and a PDF. Look at it for a moment, please!

Because I know that a stepper motor works on discrete steps, on the other hand, 3 Phase brushless works continuously, you understand me, I understood that something with PWM is also needed
[image]
[image]
[um2850-getting-started-with-the-evspin32g4-evspin32g4nh-stmicroelectronics.pdf](https://github.com/MarlinFirmware/Marlin/files/13805608/um2850-getting-started-with-the-evspin32g4-evspin32g4nh-stmicroelectronics.pdf)

EvilGremlin, 2024-01-01:
Nope, impossible. This is not motion controller at all, this is simple speed controller, completely irrelevant

closed 2024-01-01 by thisiskeithb