MarlinFirmware/Marlin #26608: I am not able to compile

WenzelHyde, 2024-01-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am really sorry if I am not in right place, but I am running out of options. I am not able make registration on reprap.org, and I am trying, googling, installing other and other libraries and VSC extensions, watching video tutorials, asking on different places and after wasted week I am not able to compile even original bugfix-2.1.x... Just because I need to do few simple changes in firmware for my Artillery printer. Could somebody please advise me what I'm doing wrong, or at least redirect me to some place where somebody will willing to help? This is output after attempt of compiling bugfix-2.1.x:

Resolving mega2560 dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing mega2560 (board: megaatmega2560; platform: atmelavr@~4.0.1; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (4.0.1) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) External (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
Converting Marlin.ino
The screen cannot be set to the number of lines and columns specified.
AssertionError: :
  File "C:\Users\venca\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\venca\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\venca\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\venca\.platformio\platforms\atmelavr\builder\main.py", line 161:
    target_elf = env.BuildProgram()
  File "C:\Users\venca\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\venca\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\venca\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\venca\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\venca\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\venca\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 339:
    env.ConvertInoToCpp()
  File "C:\Users\venca\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\venca\.platformio\penv\Lib\site-packages\platformio\builder\tools\pioino.py", line 237:
    out_file = c.convert(ino_nodes)
  File "C:\Users\venca\.platformio\penv\Lib\site-packages\platformio\builder\tools\pioino.py", line 85:
    return self.process(contents)
  File "C:\Users\venca\.platformio\penv\Lib\site-packages\platformio\builder\tools\pioino.py", line 106:
    assert self._gcc_preprocess(contents, out_file)
========================== [FAILED] Took 1.04 seconds ==========================

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:01.038
==================== 1 failed, 0 succeeded in 00:00:01.038 ====================

### Bug Timeline

New

### Expected behavior

I expected compiling firmware.

### Actual behavior

Compiling ends with one or more errors.

### Steps to Reproduce

1. Install VSC, Platformio, Python, Auto Build Marlin, etc...
2. Open bugfix-2.1.x
3. Click on Build

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Artillery Sidewinder X2

### Electronics

Stock

### LCD/Controller

Stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Uploading Config.zip…]()

thisiskeithb, 2024-01-01:
Your configs didn't upload, but the [stock Artillery Sidewinder X2 configs](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Artillery/Sidewinder%20X2) compile fine:

```prolog
Linking .pio/build/Artillery_Ruby/firmware.elf
Checking size .pio/build/Artillery_Ruby/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  39.8% (used 26052 bytes from 65536 bytes)
Flash: [=====     ]  48.1% (used 125964 bytes from 262144 bytes)
Building .pio/build/Artillery_Ruby/firmware.bin
======================================== [SUCCESS] Took 49.54 seconds ========================================

Environment     Status    Duration
--------------  --------  ------------
Artillery_Ruby  SUCCESS   00:00:49.542
======================================== 1 succeeded in 00:00:49.542 ========================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

Make sure you're using the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) and:

1. Delete the hidden `.pio` folder at the root of your Marlin project.

2. Delete the hidden `.platformio` folder from your User directory:
 
   - Windows: `%USERPROFILE%\.platformio\`
   - macOS, Unix, Unix-like: `~/.platformio/`

3. Reinstall PlatformIO.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-01-01 by thisiskeithb

WenzelHyde, 2024-01-03:
"PIO should not be using mega2560 for this board/config."

But how can I affect it? In Marlin Auto Build is only one option... I don't even know what should be there...
[image]

EvilGremlin, 2024-01-03:
atrillery_ruby env isn't mega2560 env. It use extended pin set

EvilGremlin, 2024-01-03:
You don't, ABM already set corect environment for you.

WenzelHyde, 2024-01-03:
OK, so how can I proceed? I'm using unchanged files from last bug fix, thisiskeithb already checked they shoud work... I'm sure that problem is on my VSC setup, but after dozens installs, reinstalls and reading endless similar issues I have no more ideas...

EvilGremlin, 2024-01-03:
set env manually in platformio.ini then, or run manually `platformio run -e target_env`

thisiskeithb, 2024-01-03:
Please use the resources [linked in my comment above](https://github.com/MarlinFirmware/Marlin/issues/26608#issuecomment-1873479151) for further troubleshooting since this isn't a Marlin bug.