MarlinFirmware/Marlin #26609: [BUG] Unable to compile 2.1.x for STM32F103RE_btt_USB

autonumous, 2024-01-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem is resolved, but....

### Bug Description

Unable to compile 2.1.x when configured for STM32F103RE_btt_USB (other ends are ok , eg STM32F103RE_btt)

Also  bugfix-2.1.x compiles ok,   

### Bug Timeline

Appeared after Dec 9th with 'Specify versions in INI' @4f21ace8a4415a424e191e1dde3a9846b4a13384

### Expected behavior

should be able to compile 2.1.x for STM32F103RE_btt_USB

### Actual behavior

errors
```
KeyError: '39f37d6d6a':
  File ".platformio/penv/lib/python3.11/site-packages/platformio/builder/main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File ".platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File ".platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "Desktop/Marlin-2.1.x/buildroot/share/PlatformIO/scripts/generic_create_variant.py", line 45:
    FRAMEWORK_DIR = Path(platform.get_package_dir(platform_name))
  File ".platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py", line 32:
    pkg = self.get_package(name)
  File ".platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py", line 29:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File ".platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py", line 21:
    owner=self.packages[name].get("owner"),

```

### Steps to Reproduce

1) download vanilla 2.1.x code
2) in platform.ini
default_envs = STM32F103RE_btt_USB

3) in configuration.h
MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0
SERIAL_PORT 2

compile fails 

### Version of Marlin Firmware

2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

can fix with either the following:
1) in ini/stm32-common.ini

change
platform_packages = framework-arduinoststm32@https://github.com/rhapsodyv/Arduino_Core_STM32/archive/39f37d6d6a.zip

to 
platform_packages = framework-arduinoststm32@https://github.com/rhapsodyv/Arduino_Core_STM32/archive/usb-host-msc-cdc-msc-3.zip

2) in buildroot/share/PlatformIO/scripts/generic_create_variant.py, add "39f37d6d6a" to line 36

3) update the script generic_create_variant.py to replicate the changes in bugfix-2.1.x.

thisiskeithb, 2024-01-01:
Since this is already patched in `bugfix-2.1.x`, it'll be fixed in the next `2.1.x` update / release.

closed 2024-01-01 by thisiskeithb

fmjaguar7, 2024-01-08:
hi community. i'm very new to compiling Marlin but seem to be experience this same issue. I've tried with 2.1.x branch and get this error. It seems that a week ago this fix was merged to 2.1.x if I'm reading the above correctly. Am i required to use the 2.1.x-bugfix branch instead?

thisiskeithb, 2024-01-08:
> It seems that a week ago this fix was merged to 2.1.x if I'm reading the above correctly. Am i required to use the 2.1.x-bugfix branch instead?

@fmjaguar7: You will need to use [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip).