MarlinFirmware/Marlin #26610: [makefile] avr-gcc: error: device-specs/specs-atmega1284p: No such file or directory

kardabibugergo, 2024-01-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I successfully build marlin project using VSCode PlatformIO & Marlin auto build.
>pio run -e melzi
this works pretty well

Make file is not clear in terms of what path should exists and how to set it up correctly

when trying to build with "make" command (Windows (Ubuntu bash subsystem))
I had this issue https://github.com/MarlinFirmware/Marlin/issues/2960

### Bug Timeline

new?

### Expected behavior

I expected that make file has some additional check if any of the pass is missing or incorrect and to point out what is missing

### Actual behavior

just gives you an error with no clear pass to resolve it
also if Arduino SDK and any other external tooling with some min-max version in place would help

### Steps to Reproduce

install Arduino IDE
download Marlin project 
install (sanguino https://github.com/Lauszus/Sanguino)
copy configuration.h and configuration_adv.h as described

Edit makefile
ARDUINO_INSTALL_DIR, 
AVR_TOOLS_PATH, 
ARDUINO_USER_DIR,
added EXE_EXT (important for builds with windows toolchains (avr-gcc.exe)
open C:\Windows\System32\bash.exe ~
cd to Marlin folder 
make

Compilation Error:
me@pi:/mnt/c/pi3/printer-firmwere/Marlin/Marlin$ make
Makefile:138: This GCC version 40804 is likely broken. Enabling relocation workaround.
Makefile:708: Path /mnt/c/Users/me/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/
avr-gcc: error: /mnt/c/Users/me/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/wiring.c: No such file or directory
avr-gcc: error: device-specs/specs-atmega1284p: No such file or directory
make: *** [applet/arduino/wiring.o] Error 1

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality Ender-3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13807941/configuration.zip)

Suggestion: remove makefile and relay on plarformio as much cleaner solution

EvilGremlin, 2024-01-02:
First - just stop using arduinio, it's not really supported. Despite what Scott might say. We begging to officially drop arduino for years, no one else care about it. Beside that, it's not a bug when you explicitly going out of your way and edit makefile.

kardabibugergo, 2024-01-05:
Thank you for the suggestions, I am very new to this firmware builds and likely overlooked important comments.

To keep the story short I have managed to fix it by myself 
Turns out that avr-gcc on windows had issues with file names/path I have used -v option and $(warning ) print to understand the issue.
Also, I used mingw32-make.exe and not windows (bash) make. This is because I had sdk and avr-gcc installed using windows installer.

I would like to make this final comment to share my findings because I was a bit puzzled with the build and I hope someone else could find my comment useful.

Marlin is a really great project! Thank you for the open source and for GPL license!

closed 2024-01-05 by kardabibugergo