MarlinFirmware/Marlin #26611: [BUG] MKS TINYBEE Error in serial.h

SkyHawksPa, 2024-01-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin doesn't Work.
Issus in serial.h . Compiling Error.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13809413/Configuration.zip)
Please help me!

I get this error code:

Processing mks_tinybee (platform: espressif32@2.1.0; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.10004.210126 (1.0.4)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 26 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0
|-- EEPROM @ 1.0.3
|-- ArduinoOTA @ 1.0
|-- ESPmDNS @ 1.0
|-- WiFi @ 1.0
|-- FS @ 1.0
|-- SPIFFS @ 1.0
|-- Wire @ 1.0.1
Building in release mode
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\HAL.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\HAL_SPI.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\Servo.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\Tone.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\WebSocketSerial.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\eeprom.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\i2s.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\spiffs.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\timers.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\u8g_esp32_spi.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\web.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\ESP32\wifi.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\shared\HAL.cpp.o
Compiling .pio\build\mks_tinybee\src\src\HAL\shared\MinSerial.cpp.o
In file included from Marlin/src/HAL/ESP32/../../inc/MarlinConfig.h:59:0,
                 from Marlin/src/HAL/ESP32/HAL.cpp:24:
Marlin/src/HAL/ESP32/../../inc/../core/serial.h:114:56: error: wrong number of template arguments (1, should be at least 2)
   typedef MultiSerial< REPEAT_1(NUM_SERIAL, _S_MULTI) 2> SerialOutputT;
                                                        ^
In file included from Marlin/src/HAL/ESP32/../../inc/../HAL/../HAL/ESP32/FlushableHardwareSerial.h:27:0,
                 from Marlin/src/HAL/ESP32/../../inc/../HAL/../HAL/ESP32/HAL.h:45,
                 from Marlin/src/HAL/ESP32/../../inc/../HAL/HAL.h:30,
                 from Marlin/src/HAL/ESP32/../../inc/MarlinConfig.h:33,
                 from Marlin/src/HAL/ESP32/HAL.cpp:24:
Marlin/src/HAL/ESP32/../../inc/../HAL/../HAL/ESP32/../../core/serial_hook.h:204:8: note: provided for 'template<class Serial0T, class Serial1T, unsigned char offset, unsigned char step> struct MultiSerial'
 struct MultiSerial : public SerialBase< MultiSerial< REPEAT(NUM_SERIAL, _S_NAME) offset, step > > {
        ^
In file included from Marlin/src/HAL/ESP32/../../inc/MarlinConfig.h:59:0,
                 from Marlin/src/HAL/ESP32/HAL.cpp:24:
Marlin/src/HAL/ESP32/../../inc/../core/serial.h: In function 'void SERIAL_CHAR(char, Args ...)':
[... 406 lines cut ...]
K8200

### Electronics

MKS Tinybee

### LCD/Controller

MKS 35TFT

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-01-02:
Your config is just a mess.

```
  #define Z_SAFE_HOMING_X_POINT ENDSTOPPULLUP_XMAX  // (mm) X point for Z homing
  #define Z_SAFE_HOMING_Y_POINT ENDSTOPPULLUP_YMAX  // (mm) Y point for Z homing
```

You could have  X_CENTER, X_CENTER, or X_MAX_POS, Y_MAX_POS, or X_MIN_POS, Y_MIN_POS
Or any numbers between, but not ENDSTOPPULLUP_XMAX or ENDSTOPPULLUP_YMAX

And you have  
```
#define DEFAULT_NOMINAL_FILAMENT_DIA 2,85

```
needs a decimal point, not a comma

As for serial

you have #define SERIAL_PORT_2 1

The only valid ports for this board are

0  connected to ch340, for usb serial port
2  On IO16 and IO17 on exp 1
-1 Special for ESP3D_WIFISUPPORT

But it is limited to 1 hardware uart. Ie you can only have SERIAL_PORT 0 or SERIAL_PORT 2

SERIAL_PORT_2 -1 is reserved for ESP3D_WIFISUPPORT

ellensp, 2024-01-02:
I'm leaving this open as marlin used to be able to have 
SERIAL_PORT 0 and SERIAL_PORT_2 2 without ESP3D_WIFISUPPORT and it would work.

now it does not, and I think that should be worked on

a fix to bring that back is

```DIFF
diff --git a/Marlin/src/HAL/ESP32/FlushableHardwareSerial.cpp b/Marlin/src/HAL/ESP32/FlushableHardwareSerial.cpp
index 145662215a..211cfda65c 100644
--- a/Marlin/src/HAL/ESP32/FlushableHardwareSerial.cpp
+++ b/Marlin/src/HAL/ESP32/FlushableHardwareSerial.cpp
@@ -25,5 +25,6 @@
 #include "FlushableHardwareSerial.h"
 
 Serial1Class<FlushableHardwareSerial> flushableSerial(false, 0);
+Serial1Class<FlushableHardwareSerial> flushableSerial2(false, 2);
 
 #endif
diff --git a/Marlin/src/HAL/ESP32/FlushableHardwareSerial.h b/Marlin/src/HAL/ESP32/FlushableHardwareSerial.h
index 012dda8626..ee864d08fa 100644
--- a/Marlin/src/HAL/ESP32/FlushableHardwareSerial.h
+++ b/Marlin/src/HAL/ESP32/FlushableHardwareSerial.h
@@ -32,3 +32,4 @@ public:
 };
 
 extern Serial1Class<FlushableHardwareSerial> flushableSerial;
+extern Serial1Class<FlushableHardwareSerial> flushableSerial2;
diff --git a/Marlin/src/HAL/ESP32/HAL.h b/Marlin/src/HAL/ESP32/HAL.h
index 0acb3676a2..c8a66a1faf 100644
--- a/Marlin/src/HAL/ESP32/HAL.h
+++ b/Marlin/src/HAL/ESP32/HAL.h
@@ -52,12 +52,23 @@
 
 #define MYSERIAL1 flushableSerial
 
-#if ENABLED(ESP3D_WIFISUPPORT)
-  typedef ForwardSerial1Class< decltype(Serial2Socket) > DefaultSerial1;
-  extern DefaultSerial1 MSerial0;
-  #define MYSERIAL2 MSerial0
-#elif ENABLED(WIFISUPPORT)
-  #define MYSERIAL2 webSocketSerial
+#if ANY(WIFISUPPORT, ESP3D_WIFISUPPORT)
+  #if ENABLED(ESP3D_WIFISUPPORT)
+    typedef ForwardSerial1Class< decltype(Serial2Socket) > DefaultSerial1;
+    extern DefaultSerial1 MSerial0;
+    #define MYSERIAL2 MSerial0
+  #else
+    #define MYSERIAL2 webSocketSerial
+  #endif
+#elif (SERIAL_PORT_2 == 2)
+  #define MYSERIAL2 flushableSerial2
+#endif
+
+#ifdef LCD_SERIAL_PORT
+  #if LCD_SERIAL_PORT==2
+    #define LCD_SERIAL flushableSerial2
+    #define SERIAL_GET_TX_BUFFER_FREE() LCD_SERIAL.availableForWrite()
+  #endif
 #endif
 
 #define CRITICAL_SECTION_START() portENTER_CRITICAL(&hal.spinlock)

```

SkyHawksPa, 2024-01-03:
I check the serial.h. there are curved bracket errors and in M17_M18_M19.ccp too.

ellensp, 2024-01-03:
The problems tab is not reliable. marlin is to complicated for vscode, look at real errors in Terminal tab

closed 2024-04-17 by github-actions[bot]