MarlinFirmware/Marlin #26614: [BUG] Stop working when using ft_motion. Freeze at start.

stormlyx, 2024-01-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When trying to print a test ring-tower model with changeable parameters in a Prusa slicer, printing stops before it starts, and the printer can only be stopped by resetting.

### Bug Timeline

New Bug?

### Expected behavior

Printer can print and worked properly

### Actual behavior

Fails to home when ft_motion on, and if enable ft_motion after homing in pursa slicer, then printer just stop working.

### Steps to Reproduce

Prepare g-code with ringing tower, and enable ft_motion after g28, printer don't work properly
If ft_motion enabled, printer can't home.

### Version of Marlin Firmware

2.1.x

### Printer model

ender-3

### Electronics

btt skr 1.4

### LCD/Controller

mks_mini_12864

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

staring g-code
M493 S0
M104 S[first_layer_temperature]
M140 S[first_layer_bed_temperature]
G28
M109 S[first_layer_temperature]
M190 S[first_layer_bed_temperature]
M117 ;Purge extruder
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X10.0 Y20 Z0.2 F5000.0 ; Move to start position
G1 X10.0 Y200.0 Z0.2 F1500.0 E15 ; Draw the first line
G1 X10.4 Y200 Z0.2 F5000.0 ; Move to side a little
G1 X10.4 Y20 Z0.2 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
M493 S11 D0 ; Enable ZVD Input Shaping
[archive.zip](https://github.com/MarlinFirmware/Marlin/files/13811247/archive.zip)

ellensp, 2024-01-02:
did you try the absolute latest bugfix as something very similar to this was just fixed https://github.com/MarlinFirmware/Marlin/pull/26595

stormlyx, 2024-01-02:
Yes, I try it, and nothing is changed

KALIKALI5, 2024-01-02:
How can I help!!

narno2202, 2024-01-02:
@stormlyx , I try the latest bugfix following your video step by step and I can't reproduce the error.
Regarding your GCode, It is a bad idea to change the `FT_MOTION` mode during printing even if it works in theory (ok on my printer).

stormlyx, 2024-01-02:
> @stormlyx , I try the latest bugfix following your video step by step and I can't reproduce the error. Regarding your GCode, It is a bad idea to change the `FT_MOTION` mode during printing even if it works in theory (ok on my printer).

Thank you for answer. I tried again compile new bug fix 2.1.x and it work properly. I notice what standard is zv was enabled, and maybe because of it, ft_motion did't work properly.

closed 2024-01-02 by stormlyx

thisiskeithb, 2024-01-02:
> I notice what standard is zv was enabled, and maybe because of it, ft_motion did't work properly.

We don't prevent Input Shaping & Fixed-time-based Motion Control from being compiled together & made available at the same time, but maybe more testing needs to be done around these.