MarlinFirmware/Marlin #26617: [FR] Add option to G29 Bed Leveling to skip if leveled since boot

chrisheib, 2024-01-02:
### Is your feature request related to a problem? Please describe.

While I do like to level once when starting a print session, it is quite a waste of time and sometimes reported as unnecessary wear on the printer to do it before every print in the same session. 
When slicing an object for marlin, I would need to prepare two versions of each print if I wanted to do multiple, one with and one without G29.

### Describe the feature you want

I would love if there was a flag added to G29 that would only trigger the leveling if no leveling was done this session. I believe the alternative print bed icon that shows arrows when the bed is leveled already uses such a flag.

EvilGremlin, 2024-01-02:
Just load mesh from EEPROM?

robherc, 2024-01-03:
If you enable `#define RESTORE_LEVELING_AFTER_G28` in your `configuration.h` file, you can simply manually run a G29 bed-leveling whenever you first start up your printer for a new session; then leave the sliced .gcode files with only `G28` in the startup G-code.

plampix, 2024-01-03:
G29 O

Does exactly that. You just have to make sure leveling is not enabled at boot-up. You can do this with an auto#.g file on your sd card. Combined with the earlier mentioned RESTORE_LEVELING_AFTER_G28.

EvilGremlin, 2024-01-03:
O is only for bilinear though

plampix, 2024-01-03:
> O is only for bilinear though

Ah, it's not available for Unified or manual, but it is for the rest. In that case it's a valid feature request.

Tannoo, 2024-01-03:
`#define RESTORE_LEVELING_AFTER_G28`

Run `G29` to get your leveling matrix.
Save it with `G29 S0`.
Then run `G29 A L0` at boot and it will be there.
This could be done in `CONFIGURATION_ADV.h` with `#define` STARTUP_COMMANDS "G29 A L0"`.

I agt