MarlinFirmware/Marlin #26619: [BUG] (bug summary)

DomoVlas, 2024-01-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

[{
	"resource": "/c:/Users/Laptop/Downloads/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/tft/tft_fsmc.h",
	"owner": "cpp",
	"severity": 8,
	"message": "#error \"SRAM module disabled for the STM32 framework (HAL_SRAM_MODULE_ENABLED)! Please consult the development team.\"",
	"startLineNumber": 35,
	"startColumn": 4,
	"endLineNumber": 35,
	"endColumn": 4
},{
	"resource": "/c:/Users/Laptop/Downloads/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/tft/tft_fsmc.h",
	"owner": "C/C++: IntelliSense",
	"code": "35",
	"severity": 8,
	"message": "#error directive: \"SRAM module disabled for the STM32 framework (HAL_SRAM_MODULE_ENABLED)! Please consult the development team.\"",
	"source": "C/C++",
	"startLineNumber": 35,
	"startColumn": 4,
	"endLineNumber": 35,
	"endColumn": 9
},{
	"resource": "/c:/Users/Laptop/Downloads/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/tft/tft_fsmc.h",
	"owner": "cpp",
	"severity": 8,
	"message": "'SRAM_HandleTypeDef' does not name a type; did you mean 'UART_HandleTypeDef'?",
	"startLineNumber": 60,
	"startColumn": 12,
	"endLineNumber": 60,
	"endColumn": 12
},{
	"resource": "/c:/Users/Laptop/Downloads/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/tft/tft_fsmc.h",
	"owner": "C/C++: IntelliSense",
	"code": "20",
	"severity": 8,
	"message": "identifier \"SRAM_HandleTypeDef\" is undefined",
	"source": "C/C++",
	"startLineNumber": 60,
	"startColumn": 12,
	"endLineNumber": 60,
	"endColumn": 30
},{
	"resource": "/c:/Users/Laptop/Downloads/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/tft/tft_fsmc.h",
	"owner": "cpp",
	"severity": 8,
	"message": "'FSMC_NORSRAM_DEVICE' was not declared in this scope; did you mean 'FSMC_NORSRAM_TYPEDEF'?",
	"startLineNumber": 107,
	"startColumn": 12,
	"endLineNumber": 107,
	"endColumn": 12
},{
	"resource": "/c:/Users/Laptop/Downloads/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/tft/tft_fsmc.h",
	"owner": "C/C++: IntelliSense",
	"code": "20",
	"severity": 8,
	"message": "identifier \"FSMC_NORSRAM_DEVICE\" is undefined",
	"source": "C/C++",
	"startLineNumber": 107,
	"startColumn": 12,
	"endLineNumber": 107,
	"endColumn": 31
},{
	"resource": "/c:/Users/Laptop/Downloads/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/tft/tft_fsmc.h",
	"owner": "cpp",
	"severity": 8,
[... 279 lines cut ...]
2.1.2.1

### Printer model

Anycubic Mega X

### Electronics

BTT SKR V2 + Anycubic TFT

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-01-03:
you cannot be using a ANYCUBIC_TFT35
This comes with a white flat flex cable that cannot  be plugged into a BTT SKR V2

closed 2024-01-03 by ellensp

ellensp, 2024-01-03:
This is the back of the ANYCUBIC_TFT3
[image]

ellensp, 2024-01-03:
You probably  want #define ANYCUBIC_LCD_I3MEGA
This is a serial smart screen

ellensp, 2024-01-03:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

DomoVlas, 2024-01-03:
This is what i have inside my printer. I find it as TFT screen for Mega X. Can I use it somehow with BTT SKR V2?

[image]
[image]

ellensp, 2024-01-03:
Its a DMT48320C035_06W

This is DWIN display 480x320, this is a serial device 

Electrically you just need to wire it to a serial port  power and gnd

Firmware wise. The touch screen has its own firmware files.  These files change how it talks to other firmware.
Marlin has no information on the anycubic firmware on this display.