MarlinFirmware/Marlin #26622: [BUG] Typo in esp3dlibconfig.h - MARLIN_PATH

sargonphin, 2024-01-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

There is a typo in the relative path at line 45 of the file esp3dlibconfig.h: 

`#include MARLIN_PATH(inc/MarlinConfigPre.h)`

This returns a file not found error: 

`.pio\libdeps\STM32H723ZE_btt\ESP3DLib\src\esp3dlibconfig.h:45:54: fatal error: C:/Users/x/OneDrive/3D/inc/MarlinConfigPre.h: No such file or directory`

After correcting the line I now get this and I am stuck there:

`#include MARLIN_PATH(Marlin/src/inc/MarlinConfigPre.h)`

`.pio\libdeps\STM32H723ZE_btt\ESP3DLib\src\esp3dlibconfig.h:45:54: fatal error: C:/Users/x/OneDrive/3D/Marlin/src/inc/MarlinConfigPre.h: No such file or directory`

The folder "3D" is supposed to be "3D PRINTING". It was my mistake though, I usually never put spaces in folder names for that exact reason. But I don't know how to fix that. No, renaming the folder is ABSOLUTELY NOT an option! XD This would break a considerable amount of already-configured paths for many other projects

Also as you can see I am using a BTT Octopus MAX EZ with the STM32H723ZE MCU. I am currently trying to configure the Wifi add-on board ESP32U-WROOM that I bought ages ago 

### Bug Timeline

N/A

### Expected behavior

N/A

### Actual behavior

N/A

### Steps to Reproduce

N/A

### Version of Marlin Firmware

Latest available at the time of writing

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I cannot include the configuration ZIP's for now, I will do that later if I can

thisiskeithb, 2024-01-04:
> BTT Octopus MAX EZ with the STM32H723ZE MCU. I am currently trying to configure the Wifi add-on board ESP32U-WROOM that I bought ages ago

This board is not an ESP32-based board. You simply enable the correct serial port for the WiFi addon module and not `WIFISUPPORT`/ `ESP3D_WIFISUPPORT`.

closed 2024-01-04 by thisiskeithb

sargonphin, 2024-01-04:
So what's the deal with this adapter here:

https://www.amazon.de/-/en/gp/product/B0B5ZWT3JP

It's supposed to be compatible with the board I have. Is it just not supported by Marlin at all? I can see its Wifi hotspot when I slot it into the board and such but no way to use it with Marlin so far

ellensp, 2024-01-04:
the esp32 module is a serial device for marlin, nothing more. 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

sargonphin, 2024-01-04:
I think something was not clear from the begining. My board has Wifi support as an add-in card, and has the following section in its pins, from file `Marlin\src\pins\stm32h7\pins_BTT_OCTOPUS_MAX_EZ.h`:

[image]

But it requires `WIFISUPPORT` to be enabled. This is where I am confused regarding the Wifi add-in card, because if I enable this, it complains about the initial problem I have encountered earlier.

So should this section in the pins file be removed? Or is something different expected in this port?

EvilGremlin, 2024-01-04:
These pins are not used. There was intention to untie WIFISUPPORT from platform and apply it to modules too (for potential GPIO functions) but changes weren't completed.

thisiskeithb, 2024-01-04:
Those ESP wifi addons need to be flashed with a writer like the [BTT Writer](https://biqu.equipment/products/bigtreetech-btt-writer-v1-0-module-to-esp-01s-wifi-module-expansion-3d-printer-parts-for-skr-v1-4-skr-v1-4-turbo-control-board) with firmware like [ESP3D](https://github.com/luc-github/ESP3D). Then you enable the proper serial port in Marlin and that's it. They communicate over serial.

While some non-ESP32 boards have `WIFISUPPORT` code blocks in them, they do not apply here.

sargonphin, 2024-01-04:
Okay super thanks! :) this is helpful