MarlinFirmware/Marlin #26623: [FR] Prusa like second purge after filament change

murathanaraz, 2024-01-04:
### Is your feature request related to a problem? Please describe.

Missing partial extrusion after filament change 

### Describe the feature you want

Adding a second purge after filament change before resuming like the Prusa firmware does. This way the pressure on the nozzle will be primed.

Mjankor, 2024-01-06:
Nice idea. Yes please.

thisiskeithb, 2024-01-07:
You mean `ADVANCED_PAUSE_RESUME_PRIME`?

https://github.com/MarlinFirmware/Marlin/blob/2a8c00bdeb5268b11a3bbddfc8797a3f09a92947/Marlin/Configuration_adv.h#L2912-L2912

murathanaraz, 2024-01-07:
No, that feature causes either blob or under extrusion. On Prusa firmware, when you press continue it extrudes 2cm more while it is going back to its place, you take that extra extruded filament by hand or via scissor so when it starts printing again nozzle is perfectly primed

Mjankor, 2024-01-07:
Ahhh. I was expecting the other part of Prusa's feature.
When loading filament on a Prusa, it loads, then asks whether the filament has loaded or whether it should load a bit more. This means that in the event that the filament gets stuck and doesn't pass through the extruder you've got an option to fix it. Current Marlin assumes the load was successful and goes back to printing without needing a confirmation.

thisiskeithb, 2024-01-07:
> Ahhh. I was expecting the other part of Prusa's feature.
> When loading filament on a Prusa, it loads, then asks whether the filament has loaded or whether it should load a bit more. This means that [...]

Marlin will ask if you want to purge more:

https://github.com/MarlinFirmware/Marlin/blob/2a8c00bdeb5268b11a3bbddfc8797a3f09a92947/Marlin/src/lcd/menu/menu_filament.cpp#L263

...or continue:

https://github.com/MarlinFirmware/Marlin/blob/2a8c00bdeb5268b11a3bbddfc8797a3f09a92947/Marlin/src/lcd/menu/menu_filament.cpp#L271-L271

...when changing filaments.

murathanaraz, 2024-01-07:
> Ahhh. I was expecting the other part of Prusa's feature. When loading filament on a Prusa, it loads, then asks whether the filament has loaded or whether it should load a bit more. This means that [...]

you can set the extrusion time to an extended period and you can manually stop purging when the colour is correct. IMO this is better than Prusa's sequence.

murathanaraz, 2024-01-07:
> Marlin will ask if you want to purge more:
>
> https://github.com/MarlinFirmware/Marlin/blob/2a8c00bdeb5268b11a3bbddfc8797a3f09a92947/Marlin/src/lcd/menu/menu_filament.cpp#L263
>
> ...or continue: [...]

Not the same thing. I am talking about the moment after purging more or clicking continue, there is a little more extrusion on prusa firmware so the nozzle primes without a blob

Mjankor, 2024-01-07:
Hmm. Does that occur during runouts during printing? If so, what's the requirements to activate it? I am fairly sure mine just goes straight back to printing after finishing the ADVANCED_PAUSE_PURGE_LENGTH. I'll have to test it now.

murathanaraz, 2024-01-07:
> Hmm. Does that occur during runouts during printing? If so, what's the requirements to activate it? I am fairly sure mine just goes straight back to printing after finishing the ADVANCED_PAUSE_PURGE_LENGTH. I'll have to test it now.

yes, if you set to a longer period of time you will decide where to stop purging and continue printing.

Corchoneitor, 2024-03-22:
Need this feature :).

I'm struggling with these configs right now on my Ender-3. My goal is to prevent gaps after changing filament.
First I assumed ADVANCED_PAUSE_RESUME_PRIME would do... but it creates blobs and still some under extrusion in the first layer after resume.

> No, that feature causes either blob or under extrusion. On Prusa firmware, when you press continue it extrudes 2cm more while it is going back to its place, you take that extra extruded filament by hand or via scissor so when it starts printing again nozzle is perfectly primed

I think this could solve the problem!!