MarlinFirmware/Marlin #26625: [BUG] Fan PWM working on all but one header?

sargonphin, 2024-01-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello again,

I am testing the LED and fans on the new board and I ran into problems with the PWM on FAN2 on my board (BTT Octopus Max EZ)

[image]

All fans through FAN0 to FAN6 can have properly adjusted PWM values except FAN2 which is ON/OFF. I looked everywhere in the config files and checked my wiring but couldn't find anything wrong. I have 2 identical white 12V LED strips plugged to FAN2 and FAN3, the FAN3 works perfectly fine and dims all the way through the PWM range, but FAN2 is only on-off. It is the only header with this behavior

I have not finished setting up the firmware just yet so I have not assigned this fan to anything at all. I tried to search for the pin (PA4) and the fan name (FAN2) in config and config_adv but no result. I have tried various PWM tuning settings in the config files but aside from making the fan smoother and less noisy, it didn't fix the problem I am having

I need to determine if this is a misinterpretation from Marlin or a hardware defect, from where I will get in touch with BTT

### Bug Timeline

New

### Expected behavior

All fan headers properly apply PWM to the configurable pin

### Actual behavior

All 6 but one single fan header doesn't apply PWM

### Steps to Reproduce

--

### Version of Marlin Firmware

Latest available at the time of writing

### Printer model

--

### Electronics

--

### LCD/Controller

--

### Other add-ons

--

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[CONFIG.zip](https://github.com/MarlinFirmware/Marlin/files/13825106/CONFIG.zip)

firepower9966, 2024-01-05:
#define FAN2_PIN                            PA4   // Fan2
is correct in pins_BTT_OCTOPUS_MAX_EZ.h
in Marlin-bugfix-2.1.x\Marlin\src\pins\stm32h7

You may have hardware fault,  check the mosfet on Fan2

sargonphin, 2024-01-05:
I know the pin is correct in the file, I checked it and checked wiring and MOSFET as well but no visible damage or electronic misbehavior of any kind. My guess is that the pin that Marlin tries to use is actually not correctly routed from the CPU outputs or something similar (I am not familiar with the low-level mechanics of Marlin)

firepower9966, 2024-01-05:
try swaying two of the fan pins in pins file. 
#define FAN1_PIN PA4 // Fan1
#define FAN2_PIN PA5 // Fan2

if Fan2 still fails its Software
if Fan1 Fails it probably hardware.

thisiskeithb, 2024-01-05:
I have an Octopus Max EZ I'll get hooked up this weekend to check out Fan 2.

thisiskeithb, 2024-01-05:
FAN2 (`PA4`) behavior does appear to be on/off. Tested with 24V in and VFAN2 jumper set to VIN.

I'm not sure if that's a board design bug or something within Marlin. I'd have to flash a different firmware to confirm.

Tannoo, 2024-01-06:
How does he have control of all fans and I cannot?

sargonphin, 2024-01-06:
> How does he have control of all fans and I cannot?

If you do not define a fixed number of fans in the configuration files, you can control all fans out of the box with M106 P<index> S<speed>, assuming your board is supported by Marlin. The board I use has all pins defined correctly (ish 😛) 

https://marlinfw.org/docs/gcode/M106.html

Tannoo, 2024-01-06:
Yeah... that is what I thought, but I cannot control ONE fan and was told that is by design.

I do believe the BTT SKR 3 is well supported.

sargonphin, 2024-01-06:
> Yeah... that is what I thought, but I cannot control ONE of the THREE fans and was told that is by design.
> 
> I do believe the BTT SKR 3 is well supported.

Check the board pinout to make sure you have controllable pins and not "V" something and "GND" pins. These are usually "always on" fans and are not controllable. The pins file will tell you how many and which fans you can control ^_^

robherc, 2024-01-06:
This may help here: https://github.com/bigtreetech/Octopus-Max-EZ/blob/master/Hardware%2FBIGTREETECH%20Octopus%20MAX%20EZ%20V1.0%20PinOut.png

Tannoo, 2024-01-06:
> > Yeah... that is what I thought, but I cannot control ONE of the THREE fans and was told that is by design.
> > 
> > I do believe the BTT SKR 3 is well supported.
> 
> Check the board pinout to make sure you have controllable pins and not "V" something and "GND" pins. For example MFAN on the screenshot I sent on the first post, has only VF6 and GND. VF6 is a voltage rail and is not controllable, GND is also not controllable. MFAN cannot be controlled by Marlin
> 
> These are usually "always on" fans and are not controllable. The pins file will tell you how many and which fans you can control ^_^

Yes. A dozen times. 😑

sargonphin, 2024-01-06:
Head on to the Discord server and share your config files ^_^ 

https://discord.com/invite/n5NJ59y

Tannoo, 2024-01-06:
Been there, done that.

sargonphin, 2024-01-06:
then do it again so that I can have a look :)

ellensp, 2024-01-06:
back to this actual issue

As I understand it 

For a pin to have hardware PWM functionality, the datasheet must list a alternate function of the form
TIMx_CHy for the pin concerned.

eg PA3 has TIM2_CH4, TIM5_CH4, TIM15_CH2
eg PA5 has TIM2_CH1 ,TIM8_CH1N 
There are no entry's like this for PA4

The  variant file PeripheralPins.c for this board has the same data

```
  {PA_3,       TIM2,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 4, 0)}, // TIM2_CH4
  {PA_3_ALT1,  TIM5,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM5, 4, 0)}, // TIM5_CH4
  {PA_3_ALT2,  TIM15, STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF4_TIM15, 2, 0)}, // TIM15_CH2
  {PA_5,       TIM2,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 1, 0)}, // TIM2_CH1
  {PA_5_ALT1,  TIM8,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 1, 1)}, // TIM8_CH1N

```
 Pin PA4 simply cannot do hardware PWM according to the datasheet

sargonphin, 2024-01-06:
Wtf? I guess i'm linking this to the board github

thisiskeithb, 2024-01-06:
I mentioned this on Discord, but you can use `FAN_SOFT_PWM` instead. Then adjust `SOFT_PWM_SCALE` as-needed.

thisiskeithb, 2024-01-06:
Closing since this isn't a Marlin bug.

closed 2024-01-06 by thisiskeithb