MarlinFirmware/Marlin #26629: [FR] Bring back at least 3 significant figures and multiple iterations for standard deviation during calibration.

rq3, 2024-01-04:
### Describe the feature you want

The latest bugfix only has two digits after the decimal for the standard deviation reported during calibration. Also, there used to be settings that forced a certain number of calibration iterations, with an early abort if the standard deviation was as good or better than a set value.

If these still exist, I'm darned if I can find them!

### Additional context

My delta printer has 5 micron (0.005 mm) resolution, so I really miss having better than 2 digit representation of standard deviation.

thisiskeithb, 2024-01-05:
Can you post your current config?

This is what I see after an `M48` on my Prusa Bear with `REPRAP_DISCOUNT_SMART_CONTROLLER`:

```prolog
Send: M48
M48 Z-Probe Repeatability Test
//action:notification M48 Point: 1/10
//action:notification M48 Point: 2/10
//action:notification M48 Point: 3/10
//action:notification M48 Point: 4/10
//action:notification M48 Point: 5/10
//action:notification M48 Point: 6/10
//action:notification M48 Point: 7/10
//action:notification M48 Point: 8/10
//action:notification M48 Point: 9/10
//action:notification M48 Point: 10/10
Finished!
Mean: 0.206167 Min: 0.205 Max: 0.207 Range: 0.002
Standard Deviation: 0.000553
//action:notification Deviation: 0.000553
```

`Deviation: 0.000553` on the screen in the status area.

I get similar results on my Biqu BX with `BIQU_BX_TFT70` / `TFT_COLOR_UI`.

Code supports this as well with 6 places after the decimal:

https://github.com/MarlinFirmware/Marlin/blob/a5d097abe6e8cbad3dead9086439b3c0f44c7cb2/Marlin/src/gcode/calibrate/M48.cpp#L134-L136

and

https://github.com/MarlinFirmware/Marlin/blob/a5d097abe6e8cbad3dead9086439b3c0f44c7cb2/Marlin/src/gcode/calibrate/M48.cpp#L262-L266

rq3, 2024-01-05:
M48 probe repeatability is fine, but G33 calibration seems to have lost a displayed digit and the choice of iterations that used to be available.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/13845163/configs.zip)

thisiskeithb, 2024-01-05:
> G33 calibration seems to have lost a displayed digit and the choice of iterations that used to be available.

Which version of Marlin was this available?

thisiskeithb, 2024-01-05:
Note: you can set the number of iterations with the [G33 - Delta Auto Calibration](https://marlinfw.org/docs/gcode/G033.html)'s `F` parameter:

[image]

rq3, 2024-01-05:
> > G33 calibration seems to have lost a displayed digit
> 
> Which version of Marlin was this available?

There were 3 digits after the decimal in G33 calibration for 2.1.2.1 bugfixes in June 2023.

rq3, 2024-01-05:
> Note: you can set the number of iterations with the [G33 - Delta Auto Calibration](https://marlinfw.org/docs/gcode/G033.html)'s `F` parameter:
> <img alt="image" width="732" src="https://private-user-images.githubusercontent.com/13375512/294583963-f7c55e21-d899-4c01-98b3-d58e8b462e6d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ0ODEyMTcsIm5iZiI6MTcwNDQ4MDkxNywicGF0aCI6Ii8xMzM3NTUxMi8yOTQ1ODM5NjMtZjdjNTVlMjEtZDg5OS00YzAxLTk4YjMtZDU4ZThiNDYyZTZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA1VDE4NTUxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAxODA3YTFhNThkMWY4ZDA4MzFkZjVmYzhjNWUxMGQwOWJmNzc1MmRhMmNkNGM4NjUzZjkxMjQxN2NiYzk4Y2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pActuO4IVI5H1eDIHydC3NInfeoMXI9rs_NNnQJFN44">

Yeah, but... I believe this function used to be available in the config files?

thisiskeithb, 2024-01-05:
> Yeah, but... I believe this function used to be available in the config files?

I spot checked releases all the way back to `1.1.9` and did not see a build time config option for `G33` iterations.

Feel free to check https://github.com/MarlinFirmware/Configurations#branches to confirm.

thisiskeithb, 2024-01-05:
I'm also digging through the `G33.cpp` history and I'm not seeing anything recent that shortened decimal values.

Perhaps you're thinking of `M48` results?

rq3, 2024-01-05:
> I'm also digging through the `G33.cpp` history and I'm not seeing anything recent that shortened decimal values.
> 
> Perhaps you're thinking of `M48` results?

Here are a couple of screen shots of my delta running a 2.1.2.1 bugfix from May 17, 2023. It's counting iterations, and definitely has 3 digits after the decimal for the G33 standard deviation. Does it perhaps get these from the z leveling defines in configuration.h? I notice that header is included in G33.cpp.

[image]
[image]

rq3, 2024-01-09:
I've been assiduously snipe hunting, bu still holding the bag.

Running G33 from Repetier Host works fine, and reports in 3 digits after the decimal for all coefficients, inccluding standard deviation.

Running Delta Auto Calibration from the Color_UI configuration menu works fine, but only reports 2 digits after the decimal on the touch screen for standard deviation after every iteration.

Unless you run an older commit, where it reports 3 digits on the touch screen, which is what I was after.

Obviously the data hasn't changed, since G33 over serial always reports at least 3 digits, bu the "translation" between the data, and what the 480x320 graphic TFT shows has changed, and I cannot find it. Like my previous snipe hunt, the variable is not available in the configuration files.

Simply, I need the name and location of the file that determines the number of digits displayed on the 480x320 TFT for standard deviation during a delta calibration.

My work-around is to run G33 F10 V3 over serial from Repetier Host, pick the best result, enter those results into configuration.h (for storage) AND the delta configuration on the LCD screen, save, and load.

rq3, 2024-01-16:
Binary search:
Commits up to 1fbeac7 on June 23, 2023 are OK, with standard deviations reporting as 0.000 and iterations as 01, 02, 03, etc., on the display.

Commits after that have no SD or Settings icons, until commit 574dd34 on June 27 when all hell breaks loose. During compile there is a warning about G33 truncation (!!!), but the firmware does compile. The installed firmware does load and actually gets to the info screen, and then re-boots repetitively. There are no SD or Settings Icons. 

Commit 1792608 on July 4, 2023 will compile and run, but still has no SD or Settings icons. However, it does not bootloop, so running G33 from Repetier allowed me to observe that the standard deviation was reported as 0.00, rather than the required 0.000

I suspect commit 574dd34 on June 27 is the culprit, as it touched many things. In any case, I'm within a 5 day window, and many of the commits within that window are just chron bumps.

rq3, 2024-01-23:
> I'm also digging through the `G33.cpp` history and I'm not seeing anything recent that shortened decimal values.
> 
> Perhaps you're thinking of `M48` results?

Commits up to https://github.com/MarlinFirmware/Marlin/commit/1fbeac7b5255634bb5aca209a3d3dcb6b5e43193 on June 23, 2023 are OK, with standard deviations reporting as 0.000 and iterations as 01, 02, 03, etc., on the display.

Commits after that have no SD or Settings icons, until commit https://github.com/MarlinFirmware/Marlin/commit/574dd34c49fde76e1e736b8a7e1ea579d724ff22 on June 27 when all hell breaks loose. During compile there is a warning about G33 truncation (!!!), but the firmware does compile. The installed firmware does load and actually gets to the info screen, and then re-boots repetitively. There are no SD or Settings Icons.

rq3, 2024-06-01:
This has been repaired in patched Release 2.1.2.3, but not in bugfix 2.1.x as of June 1 2024.

thisiskeithb, 2024-06-01:
> This has been repaired in patched Release 2.1.2.3, but not in bugfix 2.1.x as of June 1 2024.

Since `bugfix-2.1.x` is more than a year ahead of `2.1.2.x`, the more likely case is that whatever commit is in `bugfix-2.1.x` that changed `G33` values has not been backported to `2.1.2.x`.

Since `2.1.3` is the next expected "major" release and will be a 1:1 copy of `bugfix-2.1.x`, this will likely "break" then.

thisiskeithb, 2024-06-17:
This was output was changed in https://github.com/MarlinFirmware/Marlin/pull/24390

In `G33.cpp`, replace:

https://github.com/MarlinFirmware/Marlin/blob/a93acbd35c2ff6339a5070cb377618f73c35dc1f/Marlin/src/gcode/calibrate/G33.cpp#L637 

with:

`        MString<21> msg(F("Calibration sd:"));`:

Diff:
```diff
diff --git a/Marlin/src/gcode/calibrate/G33.cpp b/Marlin/src/gcode/calibrate/G33.cpp
index 59e0db1..12aed86 100644
--- a/Marlin/src/gcode/calibrate/G33.cpp
+++ b/Marlin/src/gcode/calibrate/G33.cpp
@@ -634,7 +634,7 @@ void GcodeSuite::G33() {
           }
         SERIAL_EOL();
 
-        MString<20> msg(F("Calibration sd:"));
+        MString<21> msg(F("Calibration sd:"));
         if (zero_std_dev_min < 1)
           msg.appendf(F("0.%03i"), (int)LROUND(zero_std_dev_min * 1000.0f));
         else
```

Tested in the Simulator with current `bugfix-2.1.x` (a93acbd):

[image]

closed 2024-07-09 by thinkyhead