MarlinFirmware/Marlin #26631: several compiling issues i need help with

debo619bawb, 2024-01-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

trying to compile marlin for a biqu b1 se plus with btt skr2 rev b board and a h2revo hotend 

### Expected behavior

to compile

### Actual behavior

does not compile

### Version of Marlin Firmware

2.1.2.1

### Printer model

biqu bb1 se plus

### Electronics

btt skr 2 rev b. h2revo hotend

### LCD/Controller

tft35 v1.0

### Bed Leveling

ABL Linear grid

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

...
[Uploading Marlin-2.1.2.1 (2).zip…]()

Marlin\src\HAL\STM32\../../inc/../core/serial.h: In function 'void SERIAL_ECHO(serial_char_t)':
Marlin\src\HAL\STM32\../../inc/../core/serial.h:61:26: error: 'MYSERIAL1' was not declared in this scope; did you mean 'MSERIAL'?
   61 |   #define _SERIAL_LEAF_1 MYSERIAL1
      |                          ^~~~~~~~~
Marlin\src\HAL\STM32\../../inc/../core/serial.h:70:25: note: in expansion of macro '_SERIAL_LEAF_1'
   70 |   #define SERIAL_LEAF_1 _SERIAL_LEAF_1
      |                         ^~~~~~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/../core/serial.h:125:31: note: in expansion of macro 'SERIAL_LEAF_1'
  125 |   #define SERIAL_IMPL         SERIAL_LEAF_1
      |                               ^~~~~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/../core/serial.h:153:44: note: in expansion of macro 'SERIAL_IMPL'
  153 | inline void SERIAL_ECHO(serial_char_t x) { SERIAL_IMPL.write(x.c); }
      |                                            ^~~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/../core/serial.h: In function 'void SERIAL_CHAR(char, Args ...)':
Marlin\src\HAL\STM32\../../inc/../core/serial.h:61:26: error: 'MYSERIAL1' was not declared in this scope; did you mean 'MSERIAL'?
   61 |   #define _SERIAL_LEAF_1 MYSERIAL1
      |                          ^~~~~~~~~
[... 757 lines cut ...]
   70 |   #define SERIAL_LEAF_1 _SERIAL_LEAF_1
      |                         ^~~~~~~~~~~~~~
Marlin\src\HAL\shared\../../core/serial.h:125:31: note: in expansion of macro 'SERIAL_LEAF_1'
  125 |   #define SERIAL_IMPL         SERIAL_LEAF_1
      |                               ^~~~~~~~~~~~~
Marlin\src\HAL\shared\../../core/serial.h:149:25: note: in expansion of macro 'SERIAL_IMPL'
  149 | void SERIAL_ECHO(T x) { SERIAL_IMPL.print(x); }
      |                         ^~~~~~~~~~~
Marlin\src\HAL\shared\../../core/serial.h: In function 'void SERIAL_ECHO(serial_char_t)':
Marlin\src\HAL\shared\../../core/serial.h:61:26: error: 'MYSERIAL1' was not declared in this scope; did you mean 'MSERIAL'?
   61 |   #define _SERIAL_LEAF_1 MYSERIAL1
      |                          ^~~~~~~~~
Marlin\src\HAL\shared\../../core/serial.h:70:25: note: in expansion of macro '_SERIAL_LEAF_1'
   70 |   #define SERIAL_LEAF_1 _SERIAL_LEAF_1
      |                         ^~~~~~~~~~~~~~
Marlin\src\HAL\shared\../../core/serial.h:125:31: note: in expansion of macro 'SERIAL_LEAF_1'
  125 |   #define SERIAL_IMPL         SERIAL_LEAF_1
      |                               ^~~~~~~~~~~~~
Marlin\src\HAL\shared\../../core/serial.h:153:44: note: in expansion of macro 'SERIAL_IMPL'
  153 | inline void SERIAL_ECHO(serial_char_t x) { SERIAL_IMPL.write(x.c); }
      |                                            ^~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_2\src\src\HAL\STM32\timers.cpp.o] Error 1
*** [.pio\build\BIGTREE_SKR_2\src\src\HAL\shared\HAL.cpp.o] Error 1
*** [.pio\build\BIGTREE_SKR_2\src\src\HAL\shared\MinSerial.cpp.o] Error 1

ellensp, 2024-01-05:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2024-01-05 by ellensp

EvilGremlin, 2024-01-05:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

* MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
* RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
* Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
* Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
* [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
* 3D Printing Discord server. Join link: https://discord.gg/3dprinters
* Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report.