MarlinFirmware/Marlin #26636: [BUG] BLTouch HS_Mode & SW_mode issue

Injineers, 2024-01-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

when using SW Mode & HS mode, I am getting errors /failures probing every other point

Seems the SW mode has something that once triggered it stows the probe. 

I re-compiled w/ SW mode disabled & probes w m48 p5  & G29 w.o any issue

attached file is w SW & HS mode enabled
[m48_HS_Mode_butBLtouchStows.txt](https://github.com/MarlinFirmware/Marlin/files/13851055/m48_HS_Mode_butBLtouchStows.txt)

### Bug Timeline

2-3 weeks ago?

### Expected behavior

when probing either w g29 or m48 it would work as expected

### Actual behavior

when running g29  
it would probe the first point, stow, move to the second point, not deploy or lower Z, then move to the next spot & probe then stow, then move to the next & not deploy or lower z, then move to the next point and probe properly , so on and so forthe

running m48 p5, it would probe 1 time & fail the second then fail out

I re-compiled w/ SW mode off & its now probing as expected

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/13851075/Configuration.zip)
[m48_HS_Mode_butBLtouchStows.txt](https://github.com/MarlinFirmware/Marlin/files/13851076/m48_HS_Mode_butBLtouchStows.txt)

### Steps to Reproduce

build latest bugfix
enable HS mode & SW mode
run a m48 p5 or a g29 & exp the issue

### Version of Marlin Firmware

bugfix from Jan 2nd

### Printer model

custom

### Electronics

SKR mini e3 v2

### LCD/Controller

Btt tft 3.5 e3 v3

### Other add-ons

Bltouch

### Bed Leveling

None

### Your Slicer

Slic3r

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

rondlh, 2024-01-06:
Are you using a real BLTouch? Or a clone (3D Touch)?
If you are using a real BLTouch, which version are you using?

Injineers, 2024-01-07:
real 3.1

enewman17, 2024-03-08:
@Injineers I too noticed this behavior, except I don't necessarily believe it's entirely with Marlin. The TFT35 screen is incredibly buggy and the code is not up to date with how Marlin 2.1 works. And since that screen is acting as a host to Marlin vs Marlin controlling the screen its introducing a lot of errors Marlin wouldn't have other wise. I also haven't had a chance to test it or install my new BL-Touch 3.1 to see if the old one is just wore out but even without using the TFT screen and using My ender 3 v2 screen I would have probing points fail but it was maybe 3 or 4 out of the hundred points on the UBL calibration vs the 10 or more that would fail using the TFT35 screen.

closed 2024-06-22 by github-actions[bot]

thisiskeithb, 2024-06-22:
I thought we resolved this on Discord. Wasn't it a wiring issue?