MarlinFirmware/Marlin #26638: [BUG] Extruder axis only goes forward (can't use retraction anymore)

PierreGillain, 2024-01-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello, i just update my Ender 3 (1.1.5 motherboard) firmware with the latest Marlin bugfix 2.1.x and i'm not able anymore to use my extruder correctly.
By hand or by gcode when i'm trying to make it go forward, it works, when i'm trying to make it go backward, it goes forward too.
The only thing i change was adding an Antclab BLTouch, so i use LCD pin 27 and +5v, GND pins for the BLTouch servo.

### Bug Timeline

After updated my motherboard to Marlin latest bugfix

### Expected behavior

I expected my extruder to go in the right direction

### Actual behavior

It moved in the same direction whatever i do

### Steps to Reproduce

-Installing a BL touch
-Configure Marlin bugfix 2.1.x to match my 3D printer
-Upload to the printer, i can start printing but every retraction move put more PLA on my print.

### Version of Marlin Firmware

Bugfix 2.1.x

### Printer model

Creality Ender-3

### Electronics

1.1.5 motherboard with Antclab BL Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[IssueGithub.zip](https://github.com/MarlinFirmware/Marlin/files/13851845/IssueGithub.zip)

ellensp, 2024-01-07:
Why do you have?
#define NUM_SERVOS 3 // Note: Servo index starts with 0 for M280-M282 commands
#define SERVO_DELAY  {300, 300, 300}

The board only has 1 servo?

Or perhaps the included pins_MELZI_CREALITY.h (which is stock)  is the wrong one?

ellensp, 2024-01-07:
Confirmed, issue is present until you remove

```
#define NUM_SERVOS 3 // Note: Servo index starts with 0 for M280-M282 commands
#define SERVO_DELAY {300, 300, 300}

```
What I suspect is happening is when the servos are being setup Marlin sets up the first one, no issues as it has a pin defined
then it tries to setup the second and third servo, the pin is not defined so it is defaulting to pin 0 (this is in conflict with the E direction pin which is also 0)

closed 2024-01-07 by ellensp

PierreGillain, 2024-01-07:
Thank you very much, it works well now.
I thought the 3 servos has to be define for something else... I dumply copy antclab manual.