MarlinFirmware/Marlin #26639: Green Confirm Tick  is to dim.

firepower9966, 2024-01-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The Confirm Tick is very dim on the Print Screen, needs to be Brighter, the green used on the status screen and file name is correct brightness and easy to see. 

### Bug Timeline

old issue ?

### Expected behavior

All Confirm ticks should be Bright Green and easy to see,

### Actual behavior

The Confirm Tick is very dim on the Print Screen

### Steps to Reproduce

Print a file and Confirm.

### Version of Marlin Firmware

 bugfix-2.1.x date 1/1/24

### Printer model

Custom 

### Electronics

MKS Robin Nano V3.1, 

### LCD/Controller

MKS TS V2

### Other add-ons

TMC2209

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13852046/Marlin.zip)

firepower9966, 2024-01-07:
[image]

thisiskeithb, 2024-01-07:
Are you actually running current `bugfix-2.1.x`? SPI TFT support is currently broken for STM32F1 and F4 and we're waiting on https://github.com/MarlinFirmware/Marlin/pull/26052.

I just flashed an MKS Robin Nano V3.1 with an MKS TS35 V2 using your config and the screen is white (which is expected until https://github.com/MarlinFirmware/Marlin/pull/26052 is merged).

As for the dim check mark, you need to rotate the encoder to highlight it and it'll turn bright green.

closed 2024-01-07 by thisiskeithb

firepower9966, 2024-01-07:
I manually added #26052 commits as tired of waiting, but it also is DIM in builds back to bugfix 2.0.9.0
I am using touch, finger or styles. regardless it should be brighter as its difficult to see.
I check out encoder use later.

thisiskeithb, 2024-01-07:
> I am using touch, finger or styles. regardless it should be brighter as its difficult to see.

Like I said, inactive UI elements are dim until highlighted/selected.

firepower9966, 2024-01-07:
I doesn't high light if using touch, it go straight to confirm / print screen, you only see it dim.

thisiskeithb, 2024-01-07:
"Too dim" is subjective here & it's obviously not too dim to select it using touch, since you can see & touch it instead of using the TS35's encoder built into the screen to move over to it.

*All* inactive elements are dim (not just the check mark).