MarlinFirmware/Marlin #26642: [FR] Make Linear Advance values for Fixed Time Motion more usable..  (say K0.6 instead of K0.000006)

ahmetcemturan, 2024-01-07:
### Is your feature request related to a problem? Please describe.

Sort of..  It is more difficult to work with these zeros behind the comma.  The serial monitor does not display it...
M493 S1 A37.00 B37.00 D0 F0.00 H0.00 P1 K0.00
When I send M493
I get:
Fixed-Time Motion enabled.
Linear Advance enabled. Gain: 0.00001
The Actual value entered is 0.000006

### Are you looking for hardware support?

No

### Describe the feature you want

It is easier for someone to switch from the old K values to the new ones (especially as the function and the variable name are the same).
A factor of 10000 would be fitting I think..
Also it would be great if the https://marlinfw.org/tools/lin_advance/k-factor.html page had an option for Fixed Time Linear Motion calibration where it would put out M493 instead of M900..
Now that FT Motion works I think that would be very nice

### Additional context

Also some clarification whether these two K-factors should or should not be used together would be nice.

narno2202, 2024-01-07:
@ahmetcemturan , as the math for linear advance are different between standard motion and FT_MOTION, no conversion is possible. I will make a PR with M493 reporting the same values as in the menu.
A little question, do you have an MKS Tiny Bee ?

ahmetcemturan, 2024-01-07:
Regarding TinyBee: Yes I have a MKS TinyBee.
And regarding FT Motion: I rather thought that one could send the x10000 value and it could be translated to the actual used value when reading the gcode...

narno2202, 2024-01-07:
Have a look at PR #26643.
I try to port `FT_MOTION` to ESP32 based motherboard. I can compile  but i can't test (no ESP32 board). If you want to test  and give feedback,  in the provided zip, you have 2 files for replacing the same files in current bugfix.  
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13854287/Marlin.zip)

ahmetcemturan, 2024-01-08:
Well this port makes the situation even worse now.  Now I can't see the actual K value at all, as the number is truncated with two decimals.   
I think I was not clear.  The problem is that the usable range of Fixed Motion Linear Advance K is around 0.0000XX   .
For my Direct drive I use a value of 0.000006   .  I am not even using the LCD Menu. (would be unusable anyway as far as I understand as it would display only two decimals.)

Regarding TinyBee: I have downloaded and patched your zip.  I will try it out as sson as possible..

ahmetcemturan, 2024-01-08:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13864645/Marlin.zip)
I compiled with the attached Configuration but I have no movement at all.
I still have no movement with M493 S0. (maybe You should open another thread for this?)

narno2202, 2024-01-08:
Thank you. You can test the bugfix branch as @thinkyhead  has pushed a slightly modified code.

ahmetcemturan, 2024-01-09:
Oddly enough:  I just compiled the latest commit and had no motion on my Robin Nano either..  So its probably not You..

ahmetcemturan, 2024-01-09:
I tried it but it won't home correctly when FT ia enabled.
Will home X, then skip Y, then homing Z

narno2202, 2024-01-09:
@ahmetcemturan , what are the results for individual homing : G28 X, G28 Y and G28 Z ? Is manual axis movement smooth and fine in X, Y and Z?

ahmetcemturan, 2024-01-10:
https://github.com/MarlinFirmware/Marlin/issues/26592

Same

narno2202, 2024-01-10:
I can't confirm the error with latest bugfix but my printer is not a corexy. As you can home with `FT_MOTION `disabled, it's not an endstop hardware problem but probably a motion problem. Have you tried to change the values of `FTM_STEPPER_FS` and/or `FTM_STEPPERCMD_BUFF_SIZE`?

ahmetcemturan, 2024-01-11:
increase or decrease?

narno2202, 2024-01-11:
Increase with `FTM_STEPPER_FS` 40 000 and `FTM_STEPPERCMD_BUFF_SIZE` 10 000 (inital tested values), if it's ok, then you can try to find the lowest values that give you the best results.

thinkyhead, 2024-01-19:
If the PR #26643 has fully addressed this issue then it can be closed.

thisiskeithb, 2024-01-19:
> If the PR #26643 has fully addressed this issue then it can be closed.

That PR made usability worse: https://github.com/MarlinFirmware/Marlin/issues/26642#issuecomment-1880401863

> Well this port makes the situation even worse now. Now I can't see the actual K value at all, as the number is truncated with two decimals.

ahmetcemturan, 2024-01-19:
It would be better if the precision had been left at 8 or evwn increas2d to 10-- Ahmet Cem Turan+90 555 455 5444

That PR made usability worse: #26642 (comment)

ahmetcemturan, 2024-01-20:
> If the value is set with M493 Gcode, the linear advance value sent is the one saved. Thz decimals are only for the menu (alphanumeric). As I said before, If others confirm the K value around 0.000001, we need to change the way it's shown in the menu with consistency between M493 and the menu.

Are you not uslng Linear Advance yourself?-- Ahmet Cem Turan+90 555 455 5444

ahmetcemturan, 2024-02-03:
> If the value is set with M493 Gcode, the linear advance value sent is the one saved. Thz decimals are only for the menu (alphanumeric). As I said before, If others confirm the K value around 0.000001, we need to change the way it's shown in the menu with consistency between M493 and the menu.

HAve You checked with others?  
@benjaminshewan  confirmed it..
(https://github.com/MarlinFirmware/Marlin/issues/26683#issuecomment-1902780976)

thisiskeithb, 2024-02-25:
Please test https://github.com/MarlinFirmware/Marlin/pull/26785 and report your findings in that PR.

closed 2024-07-15 by thinkyhead