MarlinFirmware/Marlin #26645: [FR] Update BTT STM32H7 toolchains

thurask, 2024-01-07:
### Is your feature request related to a problem? Please describe.

Attempting to build Marlin for my board (BTT SKR3 STM32H723VG) on my ARM64 laptop (Pinebook Pro running Debian, VS Code + PlatformIO + AutoBuildMarlin) errors out during the build step:

```
luser@pbp:~/git/MarlinWorkzone$ platformio run -e STM32H723VG_btt ; echo "done" >|"/tmp/ipc"
Processing STM32H723VG_btt (board: marlin_STM32H723VG; platform: ststm32@15.4.1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.90201.0
UnknownPackageError: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ ~1.90201.0' requirements for your system 'linux_aarch64'
```

This is using commit [2a8c00b](https://github.com/MarlinFirmware/Marlin/commit/2a8c00bdeb5268b11a3bbddfc8797a3f09a92947) from bugfix-2.1.x and my own Configuration.h/Configuration_adv.h. The same commit and configuration build just fine on my Windows AMD64 laptop also using VSCode + PlatformIO + AutoBuildMarlin; that I can still build working Marlin just on a different machine is why I filed this as a feature request instead of an actual bug.

### Describe the feature you want

If I remove the pinned version of platform-ststm32 in stm32h7.ini (15.4.1 at time of writing) and use the latest platform-ststm32 (17.2.0) it _does_ successfully download and install the toolchain (version 1.120301.0) for ARM64, but it runs into another issue with a dependency:

```
Compiling .pio/build/STM32H723VG_btt/FrameworkArduino/avr/dtostrf.c.o
Compiling .pio/build/STM32H723VG_btt/FrameworkArduino/board.c.o
/home/luser/.platformio/packages/framework-arduinoststm32@4.20200.221104/cores/arduino/avr/dtostrf.c: In function 'dtostrf':
/home/luser/.platformio/packages/framework-arduinoststm32@4.20200.221104/cores/arduino/avr/dtostrf.c:34:3: error: unknown type name 'uint8_t'
   34 |   uint8_t negative = 0;
      |   ^~~~~~~
/home/luser/.platformio/packages/framework-arduinoststm32@4.20200.221104/cores/arduino/avr/dtostrf.c:24:1: note: 'uint8_t' is defined in header '<stdint.h>'; did you forget to '#include <stdint.h>'?
   23 | #include <string.h>
  +++ |+#include <stdint.h>
   24 | 
*** [.pio/build/STM32H723VG_btt/FrameworkArduino/avr/dtostrf.c.o] Error 1
```

I feel that this would need attention from people more familiar with how Marlin works internally to make sure upgrading the toolchain doesn't throw any wrenches into anything, but at least it doesn't complain about the platform.

### Additional context

Other than the added ARM64 host support I think this would be similar in spirit to #26585 ?

ellensp, 2024-01-08:
you can upgrade to the following without code changes.

platform                    = ststm32@16.1.0
platform_packages           = framework-arduinoststm32@~4.20500

but I have not tested beyond compiling. 

Beyond that, requires research and code changes.

thisiskeithb, 2024-01-08:
We’d like to update *all* platforms & toolchains to latest versions, but it’s a slow process, especially for boards/variants that need code updates.

ellensp, 2024-01-11:
Want to try something?

STM32H723VG ONLY
Latest  ststm32@17.2.0 with framework-arduinoststm32@4.20701.0

It builds, but I cant test it on real hardware.

This is a very minimal config at this stage.

If this boots, try a real config see if it fin any other issues

```DIFF
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 6e4afa16c6..e0ef66e073 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -68,7 +68,7 @@
 
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define MOTHERBOARD BOARD_BTT_SKR_V3_0
 #endif
 
 /**
@@ -79,7 +79,7 @@
  *
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-#define SERIAL_PORT 0
+#define SERIAL_PORT -1
 
 /**
  * Serial Port Baud Rate
@@ -2636,7 +2636,7 @@
  * SD Card support is disabled by default. If your controller has an SD slot,
  * you must uncomment the following option or it won't work.
  */
-//#define SDSUPPORT
+#define SDSUPPORT
 
 /**
  * SD CARD: ENABLE CRC
@@ -2919,7 +2919,7 @@
 // RepRapDiscount FULL GRAPHIC Smart Controller
 // https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
 //
-//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
+#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
 
 //
 // K.3D Full Graphic Smart Controller
diff --git a/Marlin/src/HAL/STM32/sdio.cpp b/Marlin/src/HAL/STM32/sdio.cpp
index de388b8e14..dc431523d0 100644
--- a/Marlin/src/HAL/STM32/sdio.cpp
+++ b/Marlin/src/HAL/STM32/sdio.cpp
@@ -88,13 +88,13 @@ static uint32_t clock_to_divider(uint32_t clk) {
 void HAL_SD_MspInit(SD_HandleTypeDef *hsd) {
   UNUSED(hsd);
   #ifdef SDIO_FOR_STM32H7
-    pinmap_pinout(PC_12, PinMap_SD);
-    pinmap_pinout(PD_2,  PinMap_SD);
-    pinmap_pinout(PC_8,  PinMap_SD);
+    pinmap_pinout(PC_12, PinMap_SD_CK);
+    pinmap_pinout(PD_2,  PinMap_SD_CMD);
+    pinmap_pinout(PC_8,  PinMap_SD_DATA0);
     #if PINS_EXIST(SDIO_D1, SDIO_D2, SDIO_D3)  // Define D1-D3 only for 4-bit wide SDIO bus
-      pinmap_pinout(PC_9,  PinMap_SD);
-      pinmap_pinout(PC_10, PinMap_SD);
-      pinmap_pinout(PC_11, PinMap_SD);
[... 117 lines cut ...]
@@ -72,8 +72,8 @@ board                       = marlin_STM32H743VI
 #
 [STM32H723Vx_btt]
 extends                     = stm32_variant
-platform                    = ststm32@15.4.1
-platform_packages           = framework-arduinoststm32@~4.20200.220530
+platform                    = ststm32@17.2.0
+platform_packages           = framework-arduinoststm32@4.20701.0
 board_build.offset          = 0x20000
 board_upload.offset_address = 0x08020000
 build_flags                 = ${stm32_variant.build_flags}
diff --git a/platformio.ini b/platformio.ini
index 76200cbbd5..76dc0064bf 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -13,7 +13,7 @@
 [platformio]
 src_dir      = Marlin
 boards_dir   = buildroot/share/PlatformIO/boards
-default_envs = mega2560
+default_envs = STM32H723VG_btt
 include_dir  = Marlin
 extra_configs =
     Marlin/config.ini
```

thisiskeithb, 2024-01-11:
> You do get a warning "warning: .pio/build/STM32H723VG_btt/firmware.elf has a LOAD segment with RWX permissions"

See https://github.com/MarlinFirmware/Marlin/pull/26585/commits/d3f80490665d00251260746f9105074b2c374dba for more info on this.

You’d add `-Wl,--no-warn-rwx-segment` to `STM32H723VG_btt`’s `build_flags`.