MarlinFirmware/Marlin #26647: Crazy3dprint CZ-300

Thetall-1, 2024-01-08:
### Is your feature request related to a problem? Please describe.

I just got a CZ-300 from Amazon, unaware that the main webpage is no longer there. If there is anyone than could help me update the firmware it would be VERY helpful. I am new to 3d printing and just want to get the best start i can. Thank you in advance for any help you might provide.

### Describe the feature you want

I am thinking all i need to do this update is the proper or compatible config files. If i am wrong please correct me.

thisiskeithb, 2024-01-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2024-01-08 by thisiskeithb

ellensp, 2024-01-08:
Just a as follow up. 

The Configuration files only allow you to configure things that already have code in Marlin to support them.

This printer need new motherboards added, and perhaps a new displays, and in this case  a entirely new type of CPU the ATSAM4S8C.

So no your incorrect, this printer needs a lot more than just Config files to run stock Marlin.
Someone need to first add the motherboer, the display and the CPU type to marlin.

ellensp, 2024-07-24:
For reference: Although the web page is long gone, a lot of the original data it pointed to is still available
[CZ-300 Firmware (English)](https://xyzprod.blob.core.windows.net/service-center/DIYSAM4S-1.1.0v12.bin)
[SAM-BA v2.18 for Windows](https://xyzprod.blob.core.windows.net/service-center/SAM-BA%20v2.18%20for%20Windows.exe)
[CZ-300 Firmware (Japanese)](https://xyzprod.blob.core.windows.net/service-center/CZ-300%20Firmware%20(Japanese).BIN)
[Profile_Cura_V1.0.4](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300_3D_Printer_Cura_Setup_1.0.4.exe)
[Profile_Slic3r_V1.0.3](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300_3D_Printer_Slic3r_Setup_1.0.3.exe)
[Print Object(Sample)](https://xyzprod.blob.core.windows.net/document-service-center/Print%20Object(Sample).7z)
[Print Object(Fan Cover)](https://xyzprod.blob.core.windows.net/document-service-center/Print%20Object(Fan%20Cover).7z)
[USB Driver (Win 7/8/10)](https://xyzprod.blob.core.windows.net/document-service-center/USB%20Driver.7z)
[(HD23FD1X1G9)CZ-300 USER MANUAL_EU_V1.3](https://xyzprod.blob.core.windows.net/document-service-center/(HD23FD1X1G9)CZ-300%20USER%20MANUAL_EU_V1.3.pdf)
[CZ-300 USER MANUAL_EN_V1.3](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300%20USER%20MANUAL_EN_V1.3.pdf)
[CZ-300 USER MANUAL_DE_V1.3](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300%20USER%20MANUAL_DE_V1.3.pdf)
[CZ-300 USER MANUAL_ES_V1.3](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300%20USER%20MANUAL_ES_V1.3.pdf)
[CZ-300 USER MANUAL_FR_V1.3](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300%20USER%20MANUAL_FR_V1.3.pdf)
[CZ-300 USER MANUAL_IT_V1.3](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300%20USER%20MANUAL_IT_V1.3.pdf)
[CZ-300_3D_Printer_Cura_Linux.tar](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300_3D_Printer_Cura_Linux.tar.gz)
[CZ-300_3D_Printer_Cura_macOS](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300_3D_Printer_Cura_macOS.dmg)
[Profile_Cura_V1.0.6](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300_3D_Printer_Cura_Setup_1.0.6.exe)
[(HD23FD1X203) CZ-300 3D Printer User Manual_EN_V1.4](https://xyzprod.blob.core.windows.net/document-service-center/(HD23FD1X203)USER%20MANUAL_CZ-300_EN_V1.4.pdf)
[Profile_Cura_V1.0.7](https://xyzprod.blob.core.windows.net/document-service-center/CZ-300_3D_Printer_Cura_Setup_1.0.7.exe)