MarlinFirmware/Marlin #26654: [BUG] Using Actions to compile causes errors

classicrocker883, 2024-01-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Even though there is no issue/error compiling or building through VSCode > PlatformIO, when I compile using github Actions (**.github/workflows/[made up file].yml**

One error really only happens with `MPCTEMP` enabled.

`error: 'PIDTEMPBED_START' was not declared in this scope`

so another error, I would put `#define NO_CONTROLLER_CUSTOM_WIRING_WARNING` within **Conditionals_post.h**, yet the error remains. (when using an SKR Mini E3)

so this happens *ONLY* with Github Actions to build/compile.

it seems as though **Conditionals_post.h** isn't being processed before something else.

### Steps to Reproduce

1. Make a workflow that allows you to compile a configuration
2. Use an example config or make one
3. Insert a macro or enable one that is defined globally in **Conditionals_post.h**
4. View results in Actions tab

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[ExampleConfigs-ProUI-MPC.zip](https://github.com/MarlinFirmware/Marlin/files/13878051/ExampleConfigs-ProUI-MPC.zip)

thisiskeithb, 2024-01-09:
> Even though there is no issue/error compiling or building through VSCode > PlatformIO, when I compile using github Actions (.github/workflows/[made up file].yml

> One error really only happens with MPCTEMP enabled.

`MPCTEMP` is enabled in my fork and has built fine for *ages* with it enabled. I just rebased on top of the latest `bugfix-2.1.x` without issue: https://github.com/thisiskeithb/Marlin/actions/runs/7466406847/job/20317699302

---

The only warnings I get with your config when building locally (and sending it through GitHub Actions) are the following:

```prolog
Marlin/src/inc/Warnings.cpp:35:4: warning: #warning "WARNING! Disable MARLIN_DEV_MODE for the final build!" [-Wcpp]
   35 |   #warning "WARNING! Disable MARLIN_DEV_MODE for the final build!"
      |    ^~~~~~~
Marlin/src/inc/Warnings.cpp:40:8: warning: #warning "The 'long' type is the same as the 'int' type on this platform." [-Wcpp]
   40 |       #warning "The 'long' type is the same as the 'int' type on this platform."
      |        ^~~~~~~
Marlin/src/inc/Warnings.cpp:97:6: warning: #warning "Note: Auto-assigned LCD_SERIAL_PORT. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)" [-Wcpp]
   97 |     #warning "Note: Auto-assigned LCD_SERIAL_PORT. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)"
      |      ^~~~~~~
Marlin/src/inc/Warnings.cpp:807:6: warning: #warning "For ProUI the original BOOTSCREEN_TIMEOUT of 1100 is recommended." [-Wcpp]
  807 |     #warning "For ProUI the original BOOTSCREEN_TIMEOUT of 1100 is recommended."
      |      ^~~~~~~
Marlin/src/inc/Warnings.cpp:810:6: warning: #warning "For ProUI PID_AUTOTUNE_MENU and PID_EDIT_MENU is recommended for PID tuning." [-Wcpp]
  810 |     #warning "For ProUI PID_AUTOTUNE_MENU and PID_EDIT_MENU is recommended for PID tuning."
      |      ^~~~~~~
```

---

> so another error, I would put #define NO_CONTROLLER_CUSTOM_WIRING_WARNING within Conditionals_post.h, yet the error remains. (when using an SKR Mini E3)

Add `#define NO_CONTROLLER_CUSTOM_WIRING_WARNING` to your config, nowhere else.

thisiskeithb, 2024-01-09:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-01-09 by thisiskeithb

classicrocker883, 2024-01-09:
this is a false alarm, I realized the error was within the compile-configs.yml file I used for github Actions.

when searching and replacing `PIDTEMP`, it did so globally, and that included `PIDTEMPBED`. so basically I had it do true/false for `MPCTEMP`, it was commenting out both `PIDTEMP` and `PIDTEMPBED` (which I only wanted the former). that is why it only happened with that and not compiling through VSCode.