MarlinFirmware/Marlin #26655: [FR] Input shaping - reduce Cartesian bed frequency with print progress

vovodroid, 2024-01-09:
### Is your feature request related to a problem? Please describe.

With print progress mass of bed is increasing, thus reducing resonance frequency for Cartesian printers, while input shaping Y frequency remains constant. This can actually make Y axis input shaping useless.

For instance 300g of extruded filament reduces 1kg bed frequency from 30Hz to 26Hz. (30/sqrt(1.3/1))

### Describe the feature you want

If we tell to Marlin filament density and bed mass, it can calculate amount of extruded filament from print start and thus recalculate frequency during print, as frequency is inversely of square root of mass. 

[image]

robherc, 2024-01-09:
#1: this would only apply to bed-slingers, which is only a single type of cartesian printer
#2: To be completely frank, by time the printed mass is high enough to be worth adding it into IS calculations, it's going to have too much effect for the calculations to be as straightforward as you are thinking, due to changing the effective CoG of the bed, and potentially moving the new center of gravity pretty far +/- of center along the x-axis direction. (thus introducing an entire new dimension to the potential ringing occurring).

ellensp, 2024-01-09:
#3: you would need a way to know the mass as it changes.  You cant just use printed volume, Each filament type/brand/color has a different weight per length.

robherc, 2024-01-09:
oops, forgot a massive one:
#4 You would have to know the exact mass, center of gravity, and friction coefficients of the Y carriage with an empty buildplate. Otherwise, even with perfectly modelling the weight distribution of the printed model, there would be no way to be certain that any adjustments you mak would be doing more good than harm to the input-shaping's ability to neutralize the effects of resonant vibrations in the overall system.

vovodroid, 2024-01-10:
> You would have to profile your filaments every print.

I guess for each new brand/material could be good enough.

> And what about object insertion?

I don't think it could be more than nut or something like this.

I'm sure my solution is not perfect, but could it somehow improve input shaping anyway? It's too hard to implement.

ellensp, 2024-01-10:
eg of object insertion  https://youtu.be/fNQkZ7MmGio?t=215

robherc, 2024-01-10:
> It's not hard to implement
I'm sorry, but I can't understand the motivation behind this comment. making real-time changes to the tuned resonant frequency during printing, based on comically incomplete information sounds *incredibly difficult* to me... not to mention that it most likely will end up actually reducing print quality.

ellensp, 2024-01-10:
The particular example of object insertion while printing I was thinking of was 3d printed fly wheels (for self balancing robots) where a large number of ball bearings are added to give the fly wheel way more mass.  This is probably the worse case scenario for this FR

vovodroid, 2024-01-10:
> making real-time changes to the tuned resonant frequency during printing, based on comically incomplete information sounds _incredibly difficult_ to me.

As far as I understood from this #26653 discussion it's just about update frequency variable, and calculations are very simple.

> This is probably the worse case scenario for this FR

May be, it's not a silver bullet, I agree. By a way, #26653 could be useful in such case.