MarlinFirmware/Marlin #26659: [BUG] When the Software Endstops are activated the X and Y axis cannot move (Delta printer)

Orion3090, 2024-01-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm configuring Marlin bugfix-2.1.x on a SKR 1.4 Turbo for my Kossel Linear Plus, and realized that i can (through the Motion menu) move the Z axis, but any attempt to move the X or Y axis takes the extruder to the 0,0,0 position.

If i disable the Software Endstops (through the Motion menu) then i can normally move all the axis; with the Sowftware endstops activated the autoleveling runs without errors, but it only probes the 0,0 XY position, if i deactivate Software endstops the probe probes various points of the bed normally.

What i find kinda weird, is that with or without the software endstops active (through the Motion menu) the values to which i can attempt to move the X and Y axis cannot exceed the geometry i specified in the firmware, so i don't understand the point of having these software endstops anyway, is it a bug?

### Bug Timeline

First time trying with bugfix-2.1.x

### Expected behavior

With Software endstops activated, i should be able to move the extruder within the limits of the configured geometry, with software endstops disabled, i should be able to go beyond those limits.

### Actual behavior

With Software endstops activated, any attempt to move the X or Y axis moves the extruder to the 0,0,0 position, but moving the Z axis works normally; with software endstops disable i'm able to move X and Y normally but only within the limits of the geometry configured on the firmware.

### Steps to Reproduce

1.Select Auto home
2.After Auto home Free XY (through the Motion menu and Move Axis)
3. After the printer moves down the extrueder to free the XY axis Move X (or Y) axis any amount in any direction (through the Motion menu and Move Axis)
4. The printer doesn't move the extruder the desired amount on the desired axis, but moves to 0,0,0 instead

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Anycubic Kossel linear plus

### Electronics

SKR 1.4 turbo

### LCD/Controller

RepRapDiscount Smart Controller

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[this is my configuration](https://drive.google.com/file/d/1UtyWLuLn83ccUcr-9qk8rBkThbmR10ct/view?usp=drive_link).

Orion3090, 2024-01-12:
This seems to be a bug in the bugfix-2.1.x version, after migrating my configuration to 2.1.2.1 this issue seems to be solved.

closed 2024-01-12 by Orion3090

galmiklos, 2024-02-04:
Now, that this issue had been closed with "completed", does it mean this bug will not be fixed? I am having the exact same problem, and yes, I can downgrade back to 2.1.2.1, but shouldn't it be fixed?