MarlinFirmware/Marlin #26661: [BUG] MarlinSimulator Crashes in Windows

classicrocker883, 2024-01-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I get an error `strlcpy` undefined.  
in **ini/native.ini** `-DHAS_LIBBSD` makes it so `strlcpy` is not defined because of the `#ifndef HAS_LIBBSD` directive. I dont know if this was a mistake and should be `#ifdef` instead, but removing -DHAS_LIBBSD solves that issue.

however, the "SDL2" library is in '**Program Files**' and not '**C:\**', at least thats how it defaulted in my installation. but I did notice to install now it goes to **C:** for new installs for _MSYS2_ mingw64

I was able to circumvent this by copying the **SDL2** directory into **libdeps/simulator_windows/**, or change it to "Program\ FIles" in the .ini

anyway I dont know but I used to be able to get the MarlinSimulator.exe running just fine, I have an older version of this that doesn't crash. but anytime I make a new one, it will load, and subsequently crash after like 3 seconds of being open. it actually closes instead of crashing.

### Bug Timeline

Use strlcpy with buffer size (https://github.com/MarlinFirmware/Marlin/pull/26513)

### Expected behavior

[MarlinSimulator_Working.zip](https://github.com/MarlinFirmware/Marlin/files/13916581/MarlinSimulator_Working.zip)

### Actual behavior

[MarlinSimulator_Crashes-pt1.zip](https://github.com/MarlinFirmware/Marlin/files/13916620/MarlinSimulator_Crashes-pt1.zip)
[MarlinSimulator_Crashes-pt2.zip](https://github.com/MarlinFirmware/Marlin/files/13916627/MarlinSimulator_Crashes-pt2.zip)

(extract -pt2 first)

### Steps to Reproduce

1. Take a Config example, like CrealityUI for Ender3-V2
2. I used `DWIN_LCD_PROUI`
3. To create your own MarlinSimulator, start by changing in Configuration.h **MOTHERBOARD** to *BOARD_SIMULATED*, also disable `ENDSTOP_INTERRUPTS_FEATURE`
4. In platformio.ini **default_envs =** change to *simulator_windows*.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Voxelab Aquila

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-01-12:
There is no simulation for a DWIN_LCD_PROUI (or any other dwin displays in the simulator)

thisiskeithb, 2024-01-12:
Here's the simulator config: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Simulator

```c
//
// Simulator currently supports these displays. Choose one!
//
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
//#define REPRAP_DISCOUNT_SMART_CONTROLLER
//#define TFT_CLASSIC_UI
//#define TFT_COLOR_UI

// Enable parent LCD based on your selection above
#if ANY(TFT_CLASSIC_UI, TFT_COLOR_UI)
  #define TFT_GENERIC
  #define TOUCH_SCREEN
#elif ENABLED(LIGHTWEIGHT_UI)
  #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#endif
```

ellensp, 2024-01-12:
-DHAS_LIBBSD is set on environments  where strlcpy is already part of the OS so the code doesn't need the work around in the #ifndef HAS_LIBBSD  block

EvilGremlin, 2024-01-12:
bridging sim to real serial port is possible with socat, port 3 is TCP socket on port 8099

mriscoc, 2024-01-29:
> -DHAS_LIBBSD is set on environments where strlcpy is already part of the OS so the code doesn't need the work around in the #ifndef HAS_LIBBSD block

@ellensp by default in my MSYS2 installation of Windows 10 x64 libbsd seems to be missing, I added it to MSYS2 by executing:
```shell
$ pacman -S libbsd
```
The package was installed but VSCODE still cannot find **strlcpy** (I also tried with -lbsd).

By removing -DHAS_LIBBSD from the simulator_windows environment in **native.ini** your implementation is used and the simulator can be compiled and all is working correctly.

closed 2024-01-29 by ellensp