MarlinFirmware/Marlin #26671: Incorrect Scandinavic alphabets

simppakheikki, 2024-01-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Scandinavic alphabets ä and ö (and very likely å) don't appear properly. I get strange, possibly Asian letters, which replace Scandinavic alphabets. It also looks like this affects on the alphabet following the Scandinavic letter.
I already tried to search the problem but it showed to be too time consuming task since Marlin code is rather complicated.
I wonder if this has something to do with this issue [#26379 ](https://github.com/MarlinFirmware/Marlin/issues/26379).

### Expected behavior

I expect to see a finnish word "Lämpötila". 
This also affects on other words similarly.

### Actual behavior

[image]

In the image above, letters substrings "äm" and "öt" are missing from "Lämpötila". This is only one example of this problem. Other texsts having letters ä or ö has same problem in other texts.

### Steps to Reproduce

**Setup:**
OS: Windows 11 Home
ABM: Auto Build Marlin v2.1.51
PlatformIO IDE v3.3.2
IDE: VSCode v1.85.1

Firmware update has been done via SD card transferring which is typical for Ender-3 v2 printers.

### Version of Marlin Firmware

2.1.1.2

### Printer model

Creality Ender-3 v2

### Electronics

Original electronics

### LCD/Controller

DWIN display

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin settings.zip](https://github.com/MarlinFirmware/Marlin/files/13929131/Marlin.settings.zip)

EvilGremlin, 2024-01-13:
Assets for these displays don't have any accented characters, only english set. Did yo really tried bugfix? iirc code was fiixed to convert all accented characters to non-accented ones.

simppakheikki, 2024-01-13:
Yes I tried. It won't even compile without giving [missing binary operator before token "("] preprocessing errors. Those errors were even more difficult to solve.

EvilGremlin, 2024-01-13:
"missing binary operator" is becasue you used wrong config version.
Well, i guess accents were only fixed for native displays.

simppakheikki, 2024-01-13:
After some digging I finally found right config files but problem still persists. But if it is caused by the screen, there is nothing to do. I usually prefer Finnish version if it available but I didn't know it won't work for DWIN displays.

EvilGremlin, 2024-01-13:
If it's DACAI you can try swapping font.ttf, if it's DWIN/DGUS - you also can try do dive deeper and convert the font with DGUS editor.
Surefire compromise is to just swap all accented chars in lanugage file.

closed 2024-04-27 by github-actions[bot]