MarlinFirmware/Marlin #26673: [BUG] Erratic behaviour when Homing, incorrect homing on Delta printer

Orion3090, 2024-01-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I thougt i had solved this issue but i post again for it has reapeaerd, and now i understand it even less, as my printer **sometimes**, when i home works fine, homing all axis at the same time as a delta should, but **sometimes** it homes each axis individually which is a problem in deltas.

[The beaviour](https://drive.google.com/file/d/1FJfhHBLdci66rspyOp9fxB7BQQbluQjH/view?usp=drive_link) appears to be random to me, as i haven't determined if it's caused by some other variable, [i even thoght i had solved](https://github.com/MarlinFirmware/Marlin/issues/26664) it by switching to 2.1.2.1 but after trying to print, failed and then Homing it homed again the wrong way for a delta.

[Less frecuently](https://drive.google.com/file/d/1LQyQgmPqSuDHMK_bmwZzCYt-zcnQoIDq/view?usp=drive_link) one or two of the axis (again, appears to be random) do not move to the top to hit the endstop, but just jiggle in place, without any error on the printer leaving the extruder in a false home position.

I'm at a complete loss at this, maybe something to do with the configured steps for the motors, maybe it could be something with my wiring, even thogh i've checked everything on the board many times, any help with this would be really apreciated.

### Bug Timeline

First tried with bugfix-2.1.x and then with 2.1.2.1, happens with both

### Expected behavior

All axis should home at the same time, as it should with a delta.

### Actual behavior

Sometimes it homes each axis individually, which can cause crashes and damage to the printer being a delta, sometimes it homes with all 3 axis simultaneusly correctly, other times the axis doesn't go to the top.

### Steps to Reproduce

1. G28 to Home (sometimes it does it the right way, sometimes not)
2. Move down the extruder certain distance
3. G28 to Home again

Repeating this process my printer sometimes homes correctly and sometimes incorrectly homing an axis at a time, and sometimes some of the axis not moving to the top at all, just jiggling in place.

### Version of Marlin Firmware

2.1.2.1

### Printer model

Anycubic Kossel linear plus

### Electronics

SKR 1.4 turbo

### LCD/Controller

RepRapDiscount Smart Controller

### Other add-ons

TMC2209 controllers, i'm using the sensorless homing

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[This is my actual configuration](https://drive.google.com/file/d/1Epk1_2tcdVL9Nb4TFNuV8iMIZ1MDR_7e/view?usp=drive_link) with Marlin 2.1.2.1, [this one](https://drive.google.com/file/d/1Ma-o0434wloERgwVZYeuXZ_pGB26pO-U/view?usp=drive_link) is with bugfix-2.1.x

[This is a video](https://drive.google.com/file/d/1FJfhHBLdci66rspyOp9fxB7BQQbluQjH/view?usp=drive_link) of me just homing, then moving down the extruder then homing again many times, it **really** appears to be random but i know that doesn't make any sense, just haven't found why it happens; [this ](https://drive.google.com/file/d/1fp58cTnKhA9tD-0a2r86QBHZYpnYTVQo/view?usp=drive_link)is the same but with bugfix2.1 version

[This is](https://drive.google.com/file/d/1LQyQgmPqSuDHMK_bmwZzCYt-zcnQoIDq/view?usp=drive_link) what i mean when the axis doesn't move to the top and just jiggles in place.

thisiskeithb, 2024-01-13:
> i'm using the sensorless homing

Sensorless homing likely needs to be tuned for your hardware.

<hr>

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun, Repetier Host, or OctoPrint.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Orion3090, 2024-01-15:
> > i'm using the sensorless homing
> 
> Sensorless homing likely needs to be tuned for your hardware.
> 
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:
> 
> * Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
> * Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
> * Connect to your printer from host software such as Cura, Printrun, Repetier Host, or OctoPrint.
> * Send `M502` and `M500` to ensure your Configurations are applied.
> * Issue the command `M111 S247` to enable maximum logging.
> * Perform a `G28` to do your standard homing procedure.
> * Do a `G29` to probe the bed. This will also enable bed leveling.
> * Do some of the moves that revealed problems before. Take notes.
> * Copy the log output into a **.TXT file** and attach it to your next reply.
> 
> Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Thanks for your fast answer, but i ended up disabling and not using the sensorless homing, as after investigating a little bit some people don't recommend it for delta printers; with it disabled and using normal switch endstops the printer is working perfectly on Marlin 2.1.2.1.

closed 2024-01-15 by Orion3090