MarlinFirmware/Marlin #26674: [BUG] Filament change causes a filament runout trigger

PeteAlej, 2024-01-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Before I begin, I have to clarify that I am running Marlin 2.1.2 with BTT's new TFT screen firmware, on an Artillery Sidewinder X2.

So the way my printer is set up (not sure if this is the default behavior of marlin) is that each time a filament change is manually triggered (as in I manually go to the TFT screen and select the unload option) the extruder will rise a bit and then unload the filament, but the extruder head stays there until I load filament. However, since I have a filament runout sensor connected, in order to change the roll, I have to extract the filament, which causes the sensor to trigger. The problem is that when the runout sensor detect this, it rises again in order to unload the filament, which I already manually unloaded, and displays a notification on the TFT screen which prompts me to press "ok" in order for the filament to load again and then continue printing. This causes the printer to lower once the filament has been loaded but print mid-air (because it didn't lower once more to take into the account the initial raise for the filament change that I had manually executed at first). 

### Bug Timeline

It started after I updated to Marlin 2.1.2

### Expected behavior

I expected the runout sensor to be turned off just for the period of the filament change as when I was on the stick Marlin firmware I could easily execute this change and the runout sensor wouldn't trigger.

### Actual behavior

The runout sensor triggered executing a second extruder head raise, leaving the printer printing mid-air after the filament roll was changed.

### Steps to Reproduce

1. Start a normal print job.
2. Once the printer finishes bed leveling and starts printing, manually execute a filament change on the TFT screen.
3. Remove the filament from the extruder and then from the filament runout sensor. This should trigger the on-screen prompt.
4. insert the new filament and press "ok" on the prompt, and let the printer load the new filament.
5. This will cause the TFT screen to display the "printer is busy" prompt every time you try to return to the print job screen.
6. The printer will continue printing mid-air.

### Version of Marlin Firmware

2.1.2

### Printer model

Artillery Sidewinder X2

### Electronics

Stock electronics

### LCD/Controller

Model: STM32D107

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin firmware.zip](https://github.com/MarlinFirmware/Marlin/files/13929765/Marlin.firmware.zip)

thisiskeithb, 2024-01-13:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

PeteAlej, 2024-01-15:
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.
> 
> Also, ensure you’ve enabled everything in Marlin from the [Marlin Dependencies](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#marlin-dependencies) section of the TFT firmware readme.

Hi, thanks for the response, I will try with this version, however i noticed there were some sections that were not there compared to the current firmware I have installed on my printer. Should I add them or just modify the values currently in the bugfix version?

thisiskeithb, 2024-01-15:
You'll have to compare your old configs to current bugfix configs & migrate all your settings. Some things have likely been renamed and/or moved.

PeteAlej, 2024-01-15:
Ah alright, just to be sure then, there are also some parts that seem to have been removed, should I add these parts based on my current config or leave them as they are?

closed 2024-04-30 by github-actions[bot]