MarlinFirmware/Marlin #26678: [BUG] Disabling EDITABLE_STEPS_PER_UNIT will result in EEPROM validation failing on bootup

plampix, 2024-01-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When disabling EDITABLE_STEPS_PER_UNIT, on reboot, Marlin will message you EEPROM Corrupt.

### Bug Timeline

New since #26618

### Expected behavior

EEPROM contents should stay valid between reboots (after being reinitialized of course)

### Actual behavior

Every time it boots it displays EEPROM is corrupt.

### Steps to Reproduce

1. Disable EDITABLE_STEPS_PER_UNIT
2. Build & flash Marlin
3. Reinitialize EEPROM when it asks, reboot
4. Goto step 3

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality Ender-3 1.1.4 board

### Electronics

Stock

### LCD/Controller

CR10_STOCKDISPLAY

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/13930005/configs.zip)

I can't find the cause of the issue myself, unfortunately.

ellensp, 2024-01-14:
A PR has been submitted to fix this https://github.com/MarlinFirmware/Marlin/pull/26677

plampix, 2024-01-14:
Yep, #26677 fixes it.

closed 2024-01-14 by plampix