MarlinFirmware/Marlin #26679: [BUG] Setting HYBRID_THRESHOLD to 0 hangs the board at boot

sargonphin, 2024-01-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Setting `HYBRID_THRESHOLD` in Configuration_adv.h to 0 could potentially prevent boards from booting successfully after flashing, without permanent damage

Setting `HYBRID_THRESHOLD` to 0 is possible at runtime and is effectively disabling StealthChop except at full stop

### Bug Timeline

New issue

### Expected behavior

Setting this parameter to zero is produces the same outcome than at runtime

OR

`SanityCheck.h` prevents the compilation from happening

### Actual behavior

Compilation is successful but the board does not boot. Flashing is still possible, activity is visible but the firmware hangs at loading

### Steps to Reproduce

1. Enable HYBRID_THRESHOLD
2. Set any stepper threshold to zero
3. Compile & Load

### Version of Marlin Firmware

Current Bugfix

### Printer model

--

### Electronics

BTT OCtopus Max EZ

### LCD/Controller

--

### Other add-ons

--

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-01-14:
See https://github.com/MarlinFirmware/Marlin/issues/19050#issuecomment-821874710

sargonphin, 2024-01-25:
Solved

closed 2024-01-25 by sargonphin