MarlinFirmware/Marlin #26682: [FR] Add Support for the Anycubic Kobra 2 Neo Board (Tri Gorilla Gen_v4.0.0)

eroboticdude, 2024-01-14:
### Is your feature request related to a problem? Please describe.

The stock firmware for the Kobra 2 Neo is atrocious, and very difficult to use. Anycubic has not released any updates, and many features that I would like to use, such as adjusting flow rate during a print or tuning settings during a tethered print are not available at all. Using the screen at all during tethered printing isn't possible, and the printer must be restarted to use it.

### Are you looking for hardware support?

The Tri Gorilla Gen_v4.0.0 looks very similar to the other versions in the Kobra Neo and Go. It may use the same processor as well. Only visible difference is a connector.

### Describe the feature you want

Firmware support for Marlin 2.x.x would be amazing, especially since the original firmware is already based on marlin.

ellensp, 2024-01-16:
Anycubic github has this https://github.com/ANYCUBIC-3D/Kobra2_Neo 

But this is not build-able, missing environment parts.

ellensp, 2024-01-16:
Step one, getting there code to build at all. This builds, but I have no hardware to test it on 
 https://github.com/ellensp/Marlin/tree/add_kobra2_neo

AJolly, 2024-02-10:
@ellensp btw, its pretty easy to get it running on klipper. you lose the screen but otherwise works great.