MarlinFirmware/Marlin #26685: Compiling error for Polar Kinematics in Bugfix Branch

Gokuh1980, 2024-01-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Compiling error for Polar Kinematics in Bugfix Branch

[image]

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/13937879/Config.zip)

Mi interest is compile for a esp32 but i had several errors, son instead tried to compile "as is" with the atmega2560, and ramps and the result was the same, compiling errors.

### Expected behavior

I expect it to compile without errors

### Steps to Reproduce

From the bugfix branch only activate "Classic Jerk" and "Polar Kinematics"

### Version of Marlin Firmware

Marlin-bugfix-2.1.x

### Printer model

Custom Polar 3D Printer

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-01-15:
Issue is with source filtering 

Missing line in ini/features.ini

```DIFF
diff --git a/ini/features.ini b/ini/features.ini
index 8e0cb964f1..490a503bed 100644
--- a/ini/features.ini
+++ b/ini/features.ini
@@ -371,6 +371,7 @@ PRINTCOUNTER                           = build_src_filter=+<src/module/printcoun
 HAS_BED_PROBE                          = build_src_filter=+<src/module/probe.cpp> +<src/gcode/probe/G30.cpp> +<src/gcode/probe/M401_M402.cpp> +<src/gcode/probe/M851.cpp>
 IS_SCARA                               = build_src_filter=+<src/module/scara.cpp>
 HAS_SERVOS                             = build_src_filter=+<src/module/servo.cpp> +<src/gcode/control/M280.cpp>
+POLAR                                  = build_src_filter=+<src/module/polar.cpp>
 MORGAN_SCARA                           = build_src_filter=+<src/gcode/scara>
 HAS_MICROSTEPS                         = build_src_filter=+<src/gcode/control/M350_M351.cpp>
 (ESP3D_)?WIFISUPPORT                   = AsyncTCP, ESP Async WebServer

```

Gokuh1980, 2024-01-15:
Oh Nice, now if i modify features.ini and add the line you sugested it compiles OK. Thanks!!!

closed 2024-01-15 by Gokuh1980