MarlinFirmware/Marlin #26690: [FR] Restore chamber heating on power loss recovery

vovodroid, 2024-01-15:
### Is your feature request related to a problem? Please describe.

After print recover chamber heating is not restored as bed.

### Describe the feature you want

Save chamber temperature on power loss and restore it on resume.

ellensp, 2024-01-17:
Please test my PR to add this feature

vovodroid, 2024-01-17:
I don't have it, just thought about such case.

ellensp, 2024-01-17:
If you personally don't have a need for a feature, please don't waste our time requesting it!

closed 2024-01-17 by ellensp

vovodroid, 2024-01-17:
Well, I suggested that it's not personal case, but general improvement useful for a lot of people, and making Marlin better.