MarlinFirmware/Marlin #26699: [BUG] (LA on 2208 drivers not working)

Dtwpurple, 2024-01-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Tmc2208 standalone drivers do not allow for linear advanced control. 

### Steps to Reproduce

1.Enable linear advance in config
2. flash firmware
3. put hands in head after hours of messing with k values and nothing changes. 

### Version of Marlin Firmware

2.1.x bugfix

### Printer model

Ender 3 4.2.2 board. 

### Electronics

stock. 

### LCD/Controller

stock one, with the knob

### Bed Leveling

No Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/13968739/Configuration_adv.zip)

EvilGremlin, 2024-01-17:
Your config is impossible: `_DRIVER_TYPE  TMC2208` can't exist on `BOARD_CREALITY_V422`.
Also i your bowden is sloppy you might well not see any difference.
Show photos of calibration pattern, 0 to 1 with 0.01 step

thisiskeithb, 2024-01-17:
> Your config is impossible: `_DRIVER_TYPE TMC2208` can't exist on `BOARD_CREALITY_V422`.

We should probably clarify that TMCs are installed in `*_STANDALONE` mode in the warning:

https://github.com/MarlinFirmware/Marlin/blob/220060798930227fea158a81b5f83e241b7630ff/Marlin/src/inc/Warnings.cpp#L666-L668

Dtwpurple, 2024-01-17:
2208 is possible on v422. look in the config. it tells you to use that instead of the tmc2225 drivers. as those dont exist.  " Use TMC2208/TMC2208_STANDALONE for TMC2225 drivers" .  its not a bowden issue. there is no difference when setting k values. heavy bulging all the way up the tower with NO CHANGES.

Dtwpurple, 2024-01-17:
> We should probably clarify that TMCs are installed in `*_STANDALONE` mode in the warning: [...]

they are in standalone mode. and im using the right drivers as the config tell you to use 2208 instead of 2225. anytime you build marlin with ender 3 config it will shoot that error as a reminder to check them.

thisiskeithb, 2024-01-17:
If you have TMC2208s, use `TMC2208_STANDALONE`: https://github.com/MarlinFirmware/Marlin/issues/26699#issuecomment-1896856355

Dtwpurple, 2024-01-17:
> If you have TMC2208s, use `TMC2208_STANDALONE`: [#26699 (comment)](https://github.com/MarlinFirmware/Marlin/issues/26699#issuecomment-1896856355)

bruh do you guys even check the files? it is setup as tmc2208 standalone.

EvilGremlin, 2024-01-17:
We did. Its not _STANDALONE.
Print this and show photos https://marlinfw.org/tools/lin_advance/k-factor.html

thisiskeithb, 2024-01-17:
> bruh do you guys even check the files? it is setup as tmc2208 standalone.

"bruh", not when I downloaded your config:

[image]

`#define X_DRIVER_TYPE  TMC2208` is not the same as `#define X_DRIVER_TYPE  TMC2208_STANDALONE`

classicrocker883, 2024-01-20:
> As far as I'm aware for Linear Advance to work with the Creality v4.2.2 board you'd need to do a UART mod on it. 

I thought that has since been solved since LA became available for these boards. I remember there was a whole write up about the **PR**, dont remember which one it was, but AFAIK performing that mod was no longer necessary.  
[here is a link to a write up and info about the mod before LA changes in code](https://www.reddit.com/r/ender3v2/comments/n3clew/ender_3v2_422427_board_tmc_uart_mods/)

before, yes it was required to do that mod, but LA should work normally just by enabling it.

robherc, 2024-01-20:
@i-Fixit the 4.2.2 board:
#1 works great with linear advance...so this is a nonsensical bug report
#2 has soldered-down 2208s in standalone mode, so UART is not much of an option without major PCB surgery

closed 2024-01-20 by thisiskeithb