MarlinFirmware/Marlin #26705: [BUG] Release Media does not refresh menu.

chainsawwolf, 2024-01-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When selecting Release Media on an Anet A8, the menu does not update to reflect "Attach Media" unless the scroll button is pressed. The behaviour is the same as in issue 20958, except that one only addressed color UI. 

### Bug Timeline

First observed after updating to 2.1.2.1

### Expected behavior

I expect the "Release Media" menu item to update to "Attach Media" after it is pressed.

### Actual behavior

When "Release media" is selected, the label stays at "Release Media" unless another menu action (up/down arrow) is taken. If "Release Media" is selected again, the SD card is re-mounted.

### Steps to Reproduce

1> Ensure SD card is mounted.
2> Select Release Media
3> Observe no change in Release Media menu option, or Print From Media option
4> Select down arrow
5> Observe menu option change to "Attach Media", and Print From Media is removed from menu.

### Version of Marlin Firmware

2.1.2.1

### Printer model

Anet A8

### Electronics

Stock

### LCD/Controller

Stock Anet A8 LCD

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/13991643/Marlin.zip)

robherc, 2024-01-19:
Can you please test to see if this issue still exists if you use the bugfix-2.1.x branch of Marlin Firmware, with Bugfix-2.1.x configuration files as well, please (to make sure that nothing we've already done in the last couple of months has already fixed it)?

chainsawwolf, 2024-01-19:
There seem to be changes to the Z Endstop states that may impact my BLTouch settings. Z_MIN_ENDSTOP_INVERTING was changed to Z_MIN_ENDSTOP_HIT_STATE and Z_MAX_ENDSTOP_HIT_STATE. In 2.1.2.1, Z_MIN_ENDSTOP_INVERTING  had to be changed to FALSE for BLTouch. I have no idea if the hit state values need a similar swap.

robherc, 2024-01-19:
I'm sorry, I don't see the connection?

chainsawwolf, 2024-01-19:
To test the bugfix-2.1.x version, I need to build with a new bugfix config.h. Since I don't know how to configure the new values for a BLTouch, I do not want to load it on my machine and risk damage on homing.

robherc, 2024-01-19:
ok, can you please come into the support channels on the discord server, and we can work through that part there.

chainsawwolf, 2024-01-19:
If you can direct me how to get there.

robherc, 2024-01-19:
@chainsawwolf https://discord.gg/43vVEWvn

chainsawwolf, 2024-01-19:
Issue still exists in latest Bugfix version.

closed 2024-05-03 by github-actions[bot]