MarlinFirmware/Marlin #26716: [BUG] Printing starts without heating on MKS Robin Nano V1.3 (STM32F407VE)

alexgrach, 2024-01-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Kingroon KP3S with STM32F407 board (BOARD_MKS_ROBIN_NANO_V1_3_F4) printing from SD starts immediately without heating.
Setting temp from LCD menu works as expected.

Kingroon KP3S with STM32F103/GD32F303 board (BOARD_MKS_ROBIN_NANO) print from SD works correctly.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14000913/Configuration.zip)

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Kingroon KP3S

### Electronics

Stock board with STM32F407

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

HeartOfGermany, 2024-02-19:
Please provide the responsible gcode (Just the fraction you think that is not working)
Typically you use:
M109 - Wait for Hotend Temperature
M190 - Wait for Bed Temperature
M191 - Wait for Chamber Temperature
M192 - Wait for Probe temperature

Without these the heating is just active, but it will not wait for it to complete and instead starts the next command.

alexgrach, 2024-02-19:
```
;TYPE:Custom
;Kingroon KP3S
G90 ; use absolute coordinates
M83 ; extruder relative mode
M190 S55 ; wait for bed temp
M109 S205 ; wait for extruder temp
G28 ; home all
G1 Z2 F240
G1 X5 Y15 F2500
G1 Z0.4 F240
G92 E0.0
```

Works correctly with stm32f103/gd32f303 but does not work with stm32f407 on Kingroon KP3S printers.

closed 2024-06-03 by github-actions[bot]