MarlinFirmware/Marlin #26718: [BUG] IS_CARTHESIAN && DISABLED(SEGMENT_LEVELED_MOVES) wont compile error: conversion from 'XYval<unsigned char>' to non-scalar type 'XYval<signed char>' requested

Weafyr, 2024-01-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Something is wrong after this commit https://github.com/MarlinFirmware/Marlin/commit/ba08dcfb76804a912380603b3f69831ea0d1d6eb where is changed file mesh_bed_leveling.h and from this point the firmware cannot be compiled with settings IS_CARTESIAN && DISABLED(SEGMENT_LEVELED_MOVES)

I dont understand C, but my VSCode see error on line 66 in file mesh_bed_leveling.cpp - changing the line 66 from       
`xy_int8_t scel = cell_indexes(current_position), ecel = cell_indexes(destination); `
to
`xy_uint8_t scel = cell_indexes(current_position), ecel = cell_indexes(destination);`
makes compilation possible, but I won't risk uploading it to my printer as I have no idea what it does.

[image]

### Version of Marlin Firmware

not compiled, current bugfix-2.1.x

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

sargonphin, 2024-01-23:
I did the same fix on my printer for the exact same error and flashed it successfully. Although are you sure you want MBL and not mean bilinear?

Weafyr, 2024-01-23:
I'm not sure, but my printer doesn't have any probe, and in the past I used
MBL for manual bed leveling. https://marlinfw.org/docs/gcode/G029-mbl.html

sargonphin, 2024-01-23:
Ah okay then bilinear is not for you in this case. From what I understand of the function, using unsigned should not be a problem. Give it a try and see if you can successfully save a mesh and reload it from EEPROM

Weafyr, 2024-02-23:
@thinkyhead Sorry to ping you, but could you look into this? Seems so that your commit https://github.com/MarlinFirmware/Marlin/commit/ba08dcfb76804a912380603b3f69831ea0d1d6eb created this issue.

ellensp, 2024-02-26:
change
`      xy_int8_t scel = cell_indexes(current_position), ecel = cell_indexes(destination);
`to
`      xy_uint8_t scel = cell_indexes(current_position), ecel = cell_indexes(destination);
`

and 
`      const int8_t gcx = _MAX(scel.x, ecel.x), gcy = _MAX(scel.y, ecel.y);
`to
`      const uint8_t gcx = _MAX(scel.x, ecel.x), gcy = _MAX(scel.y, ecel.y);
`

Builds without issue on the simulator

Give it a try

thisiskeithb, 2024-04-05:
This will now be fixed in https://github.com/MarlinFirmware/Marlin/pull/26936 instead.

closed 2024-04-05 by thisiskeithb