MarlinFirmware/Marlin #26722: [BUG] TMC2209 not working anymore

igna09, 2024-01-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I used my printer with Marlin 2.0.5.3 and TMC2209 (X, Y, Z, E) without any issues
Today i decided to upgrade to bugfix 2.1.x porting old configurations
I noticed that homing (sensorless) was not working anymore
I executed M122 I S0 and noticed two strange behaviours:
- despite S0 flag it continue to output values instead of stopping
- current values was wrong
I tryed with M502 to restore configurations from configuration.h and configuration_adv.h but nothing changed

So I decided to downgrade to 2.0.5.3 and everything is working again

There is some breaking change (about trinamic) upgrading from 2.0.5.3 that I'm missing or is this a bug?

### Expected behavior

it should load trinamic settings (#if HAS_TRINAMIC_CONFIG) from configuration files

### Actual behavior

it load default trinamic settings

### Version of Marlin Firmware

bugfix 2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_2_0_5_3.zip](https://github.com/MarlinFirmware/Marlin/files/14027662/Configuration_2_0_5_3.zip)
[Configuration_bugfix2_1_x.zip](https://github.com/MarlinFirmware/Marlin/files/14027675/Configuration_bugfix2_1_x.zip)

igna09, 2024-01-23:
I also checked that connection is working, in both versions i can read registers correctly

phazei, 2024-01-26:
Same for me.  It was working in 2.0.5.3 then I moved to 2.1 and it's not triggering, even if I set the sensitivity to max.  I went through old versions and it seems it doesn't work from 2.0.8
Not sure if what is supposed to be set for that has changed or not.

phazei, 2024-01-28:
It took a lot of testing, but I did finally get mine working.  SKR1.4, TMC2209, BL Touch, CoreXY:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14074554/Configuration.zip)

I particularly had issues with stealthChop and hybrid_threshold.  Depending on what they were set at, the sensorless homing didn't work at all.  I had to disable them entirely, or lower my microsteps from 256 to 64.  Hopefully these files might help you.

closed 2024-05-11 by github-actions[bot]