MarlinFirmware/Marlin #26731: [BUG] STM32_UID still broken, the serial number its is prining out is garbage.

ellensp, 2024-01-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If you enable HAS_STM32_UID on a test device It now prints out a serial number, but the number is not correct

If you cut and paste the old code back in for comparison. the UID is nothing like what the old code produced

eg enable STM32_UID_SHORT_FORM so they are formatted the same as the old way
OLD           code  UUID:004D00300200C000445243F5
CURRENT code UUID:445243F5445243F5445243F5    <- note the repeating pattern

I got this to work again by replacing 

`      SERIAL_ECHO(hex_long(UID[0]), hex_long(UID[1]), hex_long(UID[2]));
`
 with 

 ```
     SERIAL_ECHO(hex_long(UID[0]));
      SERIAL_ECHO(hex_long(UID[4]));
      SERIAL_ECHO(hex_long(UID[8]));
```
This now prints  UUID:004D00300200C000445243F5 

but when you put it back into one line 

`     SERIAL_ECHO(hex_long(UID[0]), hex_long(UID[4]), hex_long(UID[8]));  
`
It gets all scrambled up again.

Same with STM32_UID_SHORT_FORM disabled

Current code

```
      SERIAL_ECHO(
        F("CEDE2A2F-"), hex_word(UID[0]), C('-'), hex_word(UID[1]), C('-'), hex_word(UID[2]), C('-'),
        hex_word(UID[3]), hex_word(UID[4]), hex_word(UID[5])
      );
```
produces  incorrect results

replacing this with 

```
     SERIAL_ECHO(F("CEDE2A2F-"), hex_word(UID[1]));
      SERIAL_ECHO(C('-'), hex_word(UID[0]));
      SERIAL_ECHO(C('-'), hex_word(UID[9]));
      SERIAL_ECHO(C('-'), hex_word(UID[8]));
      SERIAL_ECHO(hex_word(UID[17]));
      SERIAL_ECHO(hex_word(UID[16]));

```

Remember the old result wa                  004D00300200C000445243F5
and you get correct results CEDE2A2F-004D-0030-0200-C000445243F5

but if your turn it back into one line again 

```
        F("CEDE2A2F-"), hex_word(UID[1]), C('-'), hex_word(UID[0]), C('-'), hex_word(UID[9]), C('-'),
        hex_word(UID[8]), hex_word(UID[17]), hex_word(UID[16])
```
you get a mess

CEDE2A2F-43F5-43F5-43F5-43F543F543F5    <- note the repeating pattern

### Bug Timeline

feels like forever

### Expected behavior

Correct serial number should be reported

### Actual behavior

you get effectively random repeating numbers

### Steps to Reproduce

1.   #define HAS_STM32_UID 1
2. hack in the old code in M115
```
   SERIAL_ECHOPGM(" OLD UUID:");
    uint32_t *uid_address = (uint32_t*)UID_BASE;
    for (uint8_t i = 0; i < 3; ++i) {
      const uint32_t UID = uint32_t(*(uid_address));
      uid_address += 4U;
      for (int B = 24; B >= 0; B -= 8) print_hex_byte(UID >> B);
    }
```
3. compile and upload
4. send a M115 and note serial number difference

### Version of Marlin Firmware

Bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Crazy-Charles, 2024-01-26:
Even using the old way the serial number doesn't match my STM32F103 UID.
If i read the 12 bytes at location 0x1FFFF7E8 with the data width set at 32-bit, i get 05DAFF34 35325350 51186321 (the actual UID)
The results using  M115 
CEDE2A2F-FFFF-FFFF-FFFF-FFFFFFFFFFFF
CEDE2A2F-05DA-FF34-00FF-9768FFFFFFFF
 OLD UUID:05DAFF3400FF9768FFFFFFFF

00FF9768 is located at 0x1FFFF7F8
FFFFFFFF is located at 0x1FFFF808

ellensp, 2024-01-27:
Your right... its even more broken than I thought.

Documenting for anyone reading:

For my processor stm32f401RE the Device electronic signature register address is 0x1FFF7A10 
according to https://www.st.com/resource/en/reference_manual/rm0368-stm32f401xbc-and-stm32f401xde-advanced-armbased-32bit-mcus-stmicroelectronics.pdf page 839 

I connect up my stlink and get it working under platformio 

If I ask GDB to dump these 3 addresses with 
>x/3xw 0x1FFF7A10

I get

`0x1fff7a10:	0x004d0030	0x4646500a	0x20353631
`
which does not match the old code 

OLD code UUID:004D00300200C000445243F5

or the current  code 

2035-2035-2035-203520352035

or the new code I was playing with..  

Taking a closer look 

If we do a larger memory dump

```
x/16xw 0x1FFF7A10
0x1fff7a10:	0x004d0030	0x4646500a	0x20353631	0xc000fcc0
0x1fff7a20:	0x0200c000	0x87ff08a8	0x05d2f000	0x04ad039c
0x1fff7a30:	0x445243f5	0xffffffff	0x00ffffff	0xc738c639
0x1fff7a40:	0xc53ac738	0xc53ac53a	0xc639c639	0xffffffff

```
and examine the old code output    UUID:004D0030 0200C000 445243F5

first 4 bytes 004D0030  is correct
next 4 bytes 0200C000 is at  0x1fff7a20, from wrong address
next 4 bytes 445243F5 is at 0x1fff7a30, from wrong address

ellensp, 2024-01-27:
So the long working version should be

      SERIAL_ECHO(F("CEDE2A2F-"), hex_word(UID[1]));
      SERIAL_ECHO(C('-'), hex_word(UID[0]));
      SERIAL_ECHO(C('-'), hex_word(UID[3]));
      SERIAL_ECHO(C('-'), hex_word(UID[2]));
      SERIAL_ECHO(hex_word(UID[5]));
      SERIAL_ECHO(hex_word(UID[4]));

which prints CEDE2A2F-004D-0030-4646-500A20353631

Bu the single line version

      SERIAL_ECHO(
        F("CEDE2A2F-"), hex_word(UID[1]), C('-'), hex_word(UID[0]), C('-'), hex_word(UID[3]), C('-'),
        hex_word(UID[2]), hex_word(UID[5]), hex_word(UID[4])
      );

Still produces scrambled result  UUID:CEDE2A2F-3631-3631-3631-363136313631

ellensp, 2024-01-28:
got it.

It is hex_word, this returns &_hex[byte_start + 2];
but _hex is its own buffer that gets reused

so in serial_echo the parameters hex_word(UID[1]), hex_word(UID[0]), hex_word(UID[3]), hex_word(UID[2]), hex_word(UID[5]), hex_word(UID[4]) all turn into &_hex[byte_start + 2]

so when SERIAL ECHO prints them it just print out the last contents of _hex over and over.

closed 2024-01-28 by ellensp