MarlinFirmware/Marlin #26733: Issue with "chopper_timing" on the I, J and K axes in trinamic.h?

Keeneau66, 2024-01-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Simple issue. Trying to set up a 5 axis CNC configuration with trinamic drivers on the I and J axes ... are the following changes needed?

line 251 -> change "chopper_timing.toff" to "chopper_timing_I.toff"
line 265 -> change "chopper_timing.toff" to "chopper_timing_J.toff"
line 279 -> change "chopper_timing.toff" to "chopper_timing_K.toff"

I had to make these changes for my configuration to build successfully.

### Bug Timeline

I couldn't find any existing references to this

### Version of Marlin Firmware

bugfix_2.1x

### Printer model

MPCNC with additional I and J axes

### Electronics

Octopus V1.0 board

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/14068177/Configuration_files.zip)

ellensp, 2024-01-27:
this is such a simple fix, please consider creating a PR to fix it https://marlinfw.org/docs/development/getting_started_pull_requests.html

thisiskeithb, 2024-01-28:
Thanks for the heads up! I created a PR: https://github.com/MarlinFirmware/Marlin/pull/26736

closed 2024-01-28 by thisiskeithb