MarlinFirmware/Marlin #26734: [BUG] setting Y/X_HYBRID_THRESHOLD to 0 causes firmware to not load on device

phazei, 2024-01-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I was upgrading to a new version and the screen kept wigging out, after lots of testing, I finally found out it was because the X/Y hybrid_threshold were set to 0

### Bug Timeline

It worked in 2.0.5.3, then I recently updated up to 2.1.2.1, in debugging I went through many intermediate versions trying to find the issue, and it happened in all since 2.0.7

### Expected behavior

X_HYBRID_THRESHOLD should be able to be set to 0.  I was able to do this previously, that way it still stays on the LCD menu and I can adjust it from there.  Otherwise it should at least give a build error, it shouldn't just fail when loading the bin.

### Actual behavior

I put in the sdcard with the firmware, and the screen goes wonkey after a reboot.  The screen looks just like this
https://user-images.githubusercontent.com/49490265/58320401-bbcb0b00-7e45-11e9-96c8-dbd64bbd9a85.jpg
(not my picture, but a picture of an identical screen I found)

### Steps to Reproduce

```
  #define HYBRID_THRESHOLD
  #define X_HYBRID_THRESHOLD     0 //default: 100  // [mm/s]
  #define Y_HYBRID_THRESHOLD     0 //default: 100
  #define Z_HYBRID_THRESHOLD     7  //default: 3
  #define E0_HYBRID_THRESHOLD     30
```

### Version of Marlin Firmware

2.1.2.1

### Printer model

Custom CoreXY

### Electronics

SKR1.4 Turbo, 

### LCD/Controller

REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 

### Bed Leveling

ABL Linear grid

### Your Slicer

None

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14071438/Configuration.zip)

thisiskeithb, 2024-01-27:
You can't set them to 0: https://github.com/MarlinFirmware/Marlin/issues/19050#issuecomment-821874710

closed 2024-01-27 by thisiskeithb

phazei, 2024-01-27:
But they can't be commented out either, then it gives the build errors mentioned.  So there's no way to only have it enabled for one stepper.

robherc, 2024-01-27:
> But they can't be commented out either, then it gives the build errors mentioned.  So there's no way to only have it enabled for one stepper.

The point here is that setting them to 0 is illegal and causes problems. The solution is to set them to 1 or higher.

phazei, 2024-01-27:
I did try setting the x&y threshold to 1, but that seemed to stop sensorless_homing from working anymore :(

phazei, 2024-01-27:
I changed my microsteps from 256 to 64, and now hybrid_threshold isn't messing with things so it'll work at 60mm/s now!

EvilGremlin, 2024-01-27:
now turn it down to actually sensible 16 and it'll work even better and faster

sargonphin, 2024-01-27:
Also sensorless homing disables stealthchop altogether on the X and Y axis so this setting has no effect anyway ^_^

EvilGremlin, 2024-01-27:
disables for homing only

sargonphin, 2024-01-27:
Yes yes that's what I meant, thanks for correcting! 😄

phazei, 2024-01-28:
> Also sensorless homing disables stealthchop altogether on the X and Y axis so this setting has no effect anyway ^_^

Definitely can see/hear a difference when sensorless homing if stealthChop is on or off.  Sounds quite different.   There's also a big difference in the sensitivity needed, it's much more sensitive with stealthChop, it clunks the side if stealthChop is off and I haven't adjust the sensitivity.