MarlinFirmware/Marlin #26741: [FR] UBL Mesh Wizard Optimization

Tannoo, 2024-01-28:
### Is your feature request related to a problem? Please describe.

Don't think so.
Code says not to wait for the heaters. But, it waits for the hotend, then the bed.

Should be the bed, then the hotend.
Don't need to wait for the bed to reach temp. But wait for the hotend before starting the probing.

### Are you looking for hardware support?

Nope

### Describe the feature you want

Start the bed heater first.
Then start and wait for the hotend heater.

### Additional context

My solution in `ubl.cpp`:

```
  void GcodeSuite::M1004() {

    #define ALIGN_GCODE TERN(Z_STEPPER_AUTO_ALIGN, "G34\n", "")
    #define PROBE_GCODE TERN(HAS_BED_PROBE, "G29P1\nG29P3", "G29P4R")

    #if HAS_HEATED_BED
      if (parser.seenval('B')) {                        // Handle B# parameter to set Bed temp first
        const celsius_t bed_temp = parser.value_int();  // Marlin never sends itself F or K, always C
        thermalManager.wait_for_bed(false);
        thermalManager.setTargetBed(bed_temp);
      }
    #endif

    #if HAS_HOTEND
      if (parser.seenval('H')) {                          // Handle H# parameter to then set Hotend temp
        const celsius_t hotend_temp = parser.value_int(); // Marlin never sends itself F or K, always C
        thermalManager.setTargetHotend(hotend_temp, 0);
      }
    #endif
```

ellensp, 2024-01-28:
"Don't need to wait for the bed to reach temp. But wait for the hotend before starting the probing." 

This is not right.

You need the bed to heat before probing to allow the bed to thermally expand and or twist it shape, which it only does when heated

Tannoo, 2024-01-28:
Well, by the time my hotend gets to the 150, my large bed is almost 40 out of 50 deg when it start homing.

Tannoo, 2024-01-28:
The code I posted is my edited suggestion; not what it currently is.

Tannoo, 2024-01-29:
Maybe I will flip it back around but not wait for the hotend, but wait for the bed. Let them both heat up at the same time, but wait fot the bed.

Tannoo, 2024-01-29:
Maybe this could be changed to allow both heaters to heat as an option.

Just like the QUICK_HOME option.

JPT77, 2024-02-11:
Do you probe with the nozzle or with a separate probe?
If you probe with a separate probe, there should be no need to heat the hotend at all before probing?!